Ultimaker/Cura #17024: Home traverse at the end of one-at-a-time print sequence doesn't take tallest object into account

The-Balthazar, 2023-10-18:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Anycubic Kobra Max

### Reproduction steps

With the following objects in one-at-a-time print sequence
[image]
it printed the bottom right short object last.

### Actual results

On finishing print it X-homed right into the object left of it.

### Expected results

Prior to homing on completing a print, and prior to any other traverse, it should raise the Z enough to be able to clear any other objects previously printed in sequence.

### Add your .zip and screenshots here ⬇️

⬇️

GregValiant, 2023-10-19:
Thanks for the report.
The AnyCubic Kobra Max uses the standard "Ending Gcode" from the main fdmprinter definition file.  That standard definition contains a "G28 X0 Y0".  That is supposed to mean "home the X if it has lost position and home the Y if it has lost position".  If the "X0 Y0" parameters are not understood by your firmware then it it going to auto-home.  My Ender 3 Pro does not acknowledge the X0 Y0 and will always auto-home in response to G28 with or without other parameters.

From the Cura side...all the other prints on the build plate had travel moves at "MaxHeight + 5mm" but the last model does not because the print is done at that point.  It would have been fine except for the G28 in the Ending Gcode.  The author of the AnyCubic definition file didn't see fit to provide a custom Ending Gcode.

So this is a two-headed gotcha.  The Cura team will take a look at the Cura side (move up after the last model) and you need to alter your Ending Gcode.  There is no need for that G28 line.

@MariMakes and the Cura team will take a look at this.  It would be inconvenient to have the print head move to machine_height on a 450mm tall printer at the end of each print, but there is no replacement pattern for "total_bounding_box.max.z".

MariMakes, 2023-10-20:
HI @The-Balthazar,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report and thanks for tagging me @GregValiant,

I'll bring it up to the team to see what they can do to improve it. 
Fingers crossed 🤞 

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

The-Balthazar, 2023-10-20:
I believe this is an accurate enough recreation of the state of the project to recreate the issue: [CFFFP_Subspace Card.zip](https://github.com/Ultimaker/Cura/files/13057671/CFFFP_Subspace.Card.zip)

GregValiant, 2023-10-20:
This is the end of the gcode from the slice of that project:
G1 X282.532 Y14.71 E843.27193 >> last extrusion with the **Z at 4.92**
....
G0 F6000 X282.327 Y14.71 >> wipe
;TIME_ELAPSED:25266.329106
G1 F2400 E837.27193 >> last retraction
....
G28 X0 Y0

So if your printer doesn't understand the X0 Y0 it will go home at Z=4.92 and Y=14.71.  That certainly mashes into the part in the corner.
Al the other travel moves from the end of one part to the start of the next are done at Z=20.  Since AnyCubic didn't see fit to add a custom "Ending Gcode" to their definition for your printer; you got stuck with the G28 X0 Y0 line which doesn't work when printing "One at a Time".  If your printer actually understood the X0 Y0 it would not have homed, but the nozzle would have stayed into contact with the print after that last retraction.
Creality has a "present print" line in it's Ending Gcode for pretty much all their printer models.  That would cause the print head to hit a different part as it moved to Y{machine_depth} from the front of the build plate.
So I think Cura needs to move the nozzle up to that transit height (20mm in this case) after the last model is completed.  Then it wouldn't matter how the print head moved at the end because the nozzle would be above the tallest print (which in this project is 15mm).

You current workaround for this would be to check the preview and make sure that the lowest model either gets printed first, or that the lowest model is in the left front corner.

The-Balthazar, 2023-10-22:
Thanks. Yeah working around it isn't hard if you're vigilant and know it's an issue.

MariMakes, 2023-11-27:
We believe we will not work on this anytime soon and will therefore defer this issue so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker. 
Especially since there is a workaround in place if you're vigilant and know it's an issue.

We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs.  So, thanks again, and please let us know if you run into any other issues 💪