Ultimaker/Cura #17025: Slicing Error

Bruno3DR, 2023-10-18:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Ender 3 Max Neo

### Describe model location

Center

### Describe your model

Ghost

### Add your .zip here ⬇️

[Downloads.zip](https://github.com/Ultimaker/Cura/files/13034595/Downloads.zip)

GregValiant, 2023-10-19:
I can't open that zip folder.  Try again please.

Bruno3DR, 2023-10-19:
> I can't open that zip folder. Try again please.

[Downloads.zip](https://github.com/Ultimaker/Cura/files/13041490/Downloads.zip)

GregValiant, 2023-10-19:
That's it.
The "Body" model is not watertight and may be posing a problem.  I was able to slice both models without any problem so I'm unable to duplicate the behavior.
There is a known bug in 5.4.0 that has a couple of workarounds that can free up the slicing.
You can try moving the models in Cura, rotating the models, or changing the Initial Layer Height by a little bit.
Let us know if any of those worked to allow the models to slice.
[image]

Bruno3DR, 2023-10-19:
> That's it. The "Body" model is not watertight and may be posing a problem. I was able to slice both models without any problem so I'm unable to duplicate the behavior. There is a known bug in 5.4.0 [...]

I've already tried these steps on my main computer, I've even tried uninstalling and installing and adding templates that I've printed before, but the error persists with any template I put there.

GregValiant, 2023-10-19:
This sounds like the same "often reported" bug in 5.4.0.
I didn't have any trouble slicing it with my Ender active either.
I've noticed that there aren't as many "won't slice" bug reports with 5.5beta so you might want to try that and 4.13.1 is the best of the pre-arachne versions.