Ultimaker/Cura #17028: error

baz657, 2023-10-19:
### Cura Version

5.4.0

### Operating System

Widnows 11

### Printer

ender 3 pro

### Describe your model

new model

### Add your .zip here ⬇️

[CE3S1PRO_yggdrasil_in_a_circle_v1.zip](https://github.com/Ultimaker/Cura/files/13039857/CE3S1PRO_yggdrasil_in_a_circle_v1.zip)

DanWSDev, 2023-10-19:
Hi, @baz657 I looked at your file I would suggest printing one at a time and lie it down on it side to make it less likely to cause stringing between them and to fall over. 
[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic