Ultimaker/Cura #17031: WebCam when Ultimaker printer is behind protocol tunnel

PetuW, 2023-10-19:
### Cura Version

5.4.0

### Operating System

WIndows 10

### Printer

Ultimaker S5

### Reproduction steps

We have segregated OT networks where in our case Ultimaker printers are located.
[image]

For endusers they are using our IT netwrok where they have Ultimaker 5.4.0 installed to computer.

We have opened a protocol tunnel between computer-printer.
[image]

Mostly normal functions are working wine when operating printers via protocol tunnel, but when trying to open webcam via Cura 5.4.0, it sends the traffic straight to printers original IP address, when it should snd it to local host.
[image]

There is no possibilities for us to change Cura settings for this case, seems to be hard coded.
Can You pls investigate on this bug and get fixed soonests.

### Actual results

Webcam does not open.

### Expected results

Webcam will open

### Add your .zip and screenshots here ⬇️

[image]

[image]

[image]

MariMakes, 2023-10-19:
Hey @PetuW,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

The first thing I noticed is that you are not on the latest version of the firmware. 
[You can get the latest one here](https://support.makerbot.com/s/article/1667337928766)

This issue is very much outside my field of expertise 😦😖
But to me, it doesn't seem like a bug in Ultimaker Cura, because as a tester whenever I encountered 127.0.0.1, it would be an issue with the firmware instead of with Cura because it's a generic IP address.

Have you tried connecting your printers to Digital Factory? That allows you to work with your printers on different networks and access the webcam from everywhere with your account. [You can find instructions on how to do that here. ](https://support.makerbot.com/s/article/1872100986572)

Would you still like to keep this issue open and get one of the developers involved or do you agree that we can close it?

PetuW, 2023-10-19:
I might wonder that webcam issue is related to firmware of printer. Because during my test I used Wireshark to see/analyze the traffic from my client in our IT network where I have opened protocol tunnel to printer. Basic thing works but when opening webcam it will try to connect with printers IP instead of to be used the local host address.

MariMakes, 2023-10-20:
I'll bring it up to the team to see what they can do to helpit.
But I expect that if it's not a quick fix it will not get a high priority to implement, because we also have a digital factory solution available. 

Fingers crossed 🤞

PetuW, 2023-10-31:
Well, the issue her is that as we have in our company physically segragated / separated our network infra structure, and based on this 3D printers are in OT network where based on ZeroTrust all needed communications have to be opened in our firewalls. And also as our operators commented, digital factory solution can not be used when we have several ultimaters with different filaments and setups, so they have to be as independent printers and queues. But as I tryed to expalin earlier, when we have analyzed via Wireshark the traffic when opened the protocol tunneln from our IT computer to OT printer, Ultimaker software from IT computer can get done all other than web cam printer, seems to be that software and/or printer does not understand the protocol tunneln here between where targer OT IP and port is forwarded via IT clients local address+port.

MariMakes, 2024-01-11:
Hi PetuW, 

Apologies for the late reply. Your issue has been discussed with the team but we believe we will not work on this anytime soon and will therefore defer this issue so we can focus on the most requested and needed features and bug fixes from our community and from Ultimaker

We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs.  So, thanks again, and please let us know if you run into any other issues 💪 

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.