Ultimaker/Cura #17032: Separate z-hop setting for steep overhang places.

ps1x, 2023-10-19:
### Is your feature request related to a problem?

When printing tall features that top with steep overhangs they fall of because these overhang parts tend to curl, printer hits it with nozzle and Baaam, sad story for printing large model.

### Describe the solution you'd like

I think it would be possible to detect these places with steep overhang and override z-hop value to something higher value.  This way print time will not increase that drammaticaly because normally you dont need z-hop at all, yet print will not fail because of colliding with curled overhang.

### Describe alternatives you've considered

The only alternative is to use higher z-hop for entire print, but that icreases print time and also somewhat affects quality for printers with backlash in z-nut.

### Affected users and/or printers

Might affect any printer that prints curly materials, in my case that was Ender 3 pro printing PLA.

GregValiant, 2023-10-20:
Timing is everything.
I have a new post processor in the works to address this.
Unzip and put "AlterZHops.py" into your Configuration folder in the "scripts" folder.  It will be available under Extensions / Post Processing / Modify Gcode.
You can enter ranges of layers and a new hop height and the zhops in the range will be altered.
If you set the hop height to 0 the zhops in the range simply won't do anything (but the gcode lines are not removed).
[AlterZhops.zip](https://github.com/Ultimaker/Cura/files/13049370/AlterZhops.zip)

ps1x, 2023-10-20:
@GregValiant thats really cool, thank you. Do you think it is possible to implement z-hop feature only for specific part of layer where curling may occur?

GregValiant, 2023-10-20:
That isn't technically possible.
You can enter a single layer, or ranges of layers, but can't split things within a layer.