Ultimaker/Cura #17034: The New Kobra 2 Max Locks Up.

rjack37, 2023-10-19:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Anycubic Kobra 2 Max

### Reproduction steps

Just run it.

### Actual results

The Kobra 2 Max will lock after about 6.2 hours of printing and won't work till you reboot it by turning it off then on again. I have been working with customer service at Anycubic and they can't seem to fix the problem except to change from Anycubic to PursaSlicer. When I go back to using Ultimaker Cura 4.12 everything works fine.

### Expected results

It should print with no problem.

### Add your .zip and screenshots here ⬇️

I'm not sure if its a glitch with Cura or with Anycubic but you might want to look into it since it doesn't do it with the older versions.

GregValiant, 2023-10-20:
Thanks for the report.
Are you printing via the SD card or some other way (Octoprint, USB, etc.).
Zip a "bad" gcode file and post the zip folder here.  Without something to look at there really can't be an investigation.  This behavior has not been reported by any other Kobra 2 Max owners.

zeroku3, 2023-10-31:
Try to update!!:

manual:
https://cloud-universe.anycubic.com/w/p/helpCenter/#/pages/pageDetail/pageDetails?id=771&sourceId=1&searchWord=Kobra%202%20Series%20V2.3.5%20Firmware%20Update&isShowShare=true&language=US&channelId=%206

firmware: https://drive.google.com/file/d/1aAax4XkzsjmBcNSmKSevUXDGT6EsfpXp/view?usp=drive_link

GregValiant, 2023-10-31:
Can you point to something that Cura is doing wrong?  UltiMaker is not, and can not be responsible for AnyCubic firmware.

rjack37, 2023-10-31:
Greg,I worked with Anycubic and got settings for Cura that work but still haven't been able to get it to z-hop correctly, Maybe the firmware update will fix that. I wish Cura had a setting where you could z-hop above and below a certain point or be able to Modify settings for overlaps in the modify section but I guess they already got that request from other people.  I'm in the middle of a 2 day print and if it messes up again in about 2 hours, I'll load the new firm ware.  Thank you for sending it and it is very much appreciated.

 
 

  

|  | Virus-free.www.avg.com |

rjack37, 2023-10-31:
Hey,The new version of Cura has a way to z-hop only over certain parts! Glad they did that! 

 
 

  

|  | Virus-free.www.avg.com |

zeroku3, 2023-10-31:
How can I configure the printer on the priest, if it does not have a profile?

GregValiant, 2023-10-31:
Z-hops are just a Z move of course.  There really shouldn't be any issues with it.  I've submitted a post processor to alter Z-hops because a lot of prints only need them in certain layers or ranges of layers.  The post doesn't get rid of them completely, just sets a new hop height which can be "0".  In the case of a "0" hop height the Z-hop lines don't do anything but take up space.
Unzip the file and put "AlterZhops.py" into the "scripts" folder within your Configuration folder.  It will be available with the rest of the post processors.

[AlterZhops.zip](https://github.com/Ultimaker/Cura/files/13220158/AlterZhops.zip)

zeroku3, 2023-11-05:
The same thing happens to me, I print a large piece and after several hours the machine stops, freezes, with the screen off. It is updated to version 2.3.5 and the same thing has happened to me laminating with Cura 5.5 or laminating with Anycubid Slicer!!!
I need to print pieces for two days!!! Any ideas?

rjack37, 2023-11-05:
Download and revert  Back to Cura 4.12, it works just fine. I have heard Cura 5.2 will work too but I know 4.12 works well.  I spent about a week working with AnyCubic tech Support and on the Anycubic web site, they uploaded updated new and improved Cura profile settings that take a while to dial in and to tweak to do what I need it to do, At first seemed to work but I think I'm starting to have lamination problems with those.  I'm not sure if its a glitch on the lamination part but I think it has more to do with the rush to get the Kobra2 Max out the door. But if your pushed for time, I would just go to 4.12 and the profile you already developed should work. Hope this helps, and if it does, let me know or if you need a copy of 4.12 I'll send you a copy I downloaded off the web.  

Ron

zeroku3, 2023-11-05:
I'm going to try version 4.12. but I don't understand, because the same thing happens to me with anycubic slicer.... Will we have to wait for a new firmare update?
thanks my friend

rjack37, 2023-11-05:
2.3.5 is suppose to be the latest and greatest. Anycubic slicer is not working? Is PursaSlicer working? One person said that Tech Support told him that after he loads the the file onto the printer, to pull the USB and run the file from the Printer memory instead of from the USB and he said it worked for him.  If Pursa isn't working, either, I would reload my firmware and go from there.  Sorry I can't be of anymore help.

zeroku3, 2023-11-06:
Thank you very much, with cure 4.12 it works!! but of course, it is a much worse laminator than its latest versions...
The same model, in cure 4.12 takes me 12 hours and 48 minutes, and in cure 5.5 (with all the same options) 10 hours!

I hope Anycubic solves it quickly... this doesn't make much sense...!

Thanks ;) We'll see any news!

rjack37, 2023-11-06:
Since the Kobra 2 can print at a faster rate, can you increase the speed to make up the difference? Also, have you tried SuperSlicer to see if that works? https://github.com/supermerill/SuperSlicer/releases/download/2.4.58.5/SuperSlicer_2.4.58.5_win64_220924.zip Its based on PursaSlicer but its different.  I just can't use it because of the cross directional printing of the layers (for strength), leaves the lower layers showing through on my prints.

The folks at Anycubic have produced newer profiles that work with Cura 5.4 and 5.5 https://drive.google.com/file/d/170L5J1IZMJ6ZVOkNt4dI0MkcvKeBq_f2/view They just take a while to dial in but after tweaking them to what I need, are running all the way through and I'm using all of the settings and options I normally use. I use ASA but run the ABS profile and adjust the settings for ASA.
I think Anycubic has to come up with a firmware update soon because its happening quite a bit on the other boards I watch. PLA seems to not have a problem but I can't use PLA because of the temperature limits and outside humidity absorption of the PLA filament. I'm printing front Grilles for 1986 Camaro Sport Coupes that need to be able to handle the heat and humidity so its PETG and ASA for me. My prints take 4 days normally but the Kobra 2 does them in 2 days. I hope this helps.
Ron

zeroku3, 2023-11-06:
I use the cure profiles that anycubic has in version 5.5, but it fails, after several hours of work the machine freezes, stops and not even the screen responds. The same thing happens to me with the anycubic slicer, the latest version already supports the kobra 2 max, but it gives me the same error.
At the moment with cure 4.12 I have had 10 hours of printing, record....
what a pity :(

rjack37, 2023-11-06:
Strange that the Kobra Max, not the Kobra 2 Max works just fine with Cura 5.5 but the Kobra 2 Max freezes after 5 hours 23 minutes.  I'm starting to think it has something to do with Anycubic trying to integrate the Cloud, WiFi and Camera into their firmware because the Camera port doesn't work and their Wifi app doesn't control the printer either, it only notifies you when you stop a print or when it finishes one. I hope AnyCubic fixes this soon.  When I sent AnyCubic Tech Support my print file and talked to them they said there was a piece of code in the G-Code string their printer couldn't recognize and caused it to freeze but if that were true, why would they upload new profiles as a work around and why does it do the same thing with their slicer which is based on PursaSlicer?  Anyway, congrats on 10 hours and going strong!

rjack37, 2023-11-06:
> Let's see if we find out what's going on or if they update the machine, because it has other errors, for example if you link it to the Windows app (anycubic slicer) to be able to send the prints over the network, it no longer lets you link it to the mobile app (android/ ios) and it doesn't make sense...
>
> Although I am more worried about not being able to use Cura 5.5 and later, which have many more options and improvements than version 4.12, obviously!
>
> Regards my friend

They are probably working on it now unless some of the machines shipped out with a bad batch of processors, in which case they should send out replacements for the bad chips if they can be replaced.  If not then Its a new mother board replacement which we shouldn't have to do but they are in China so were kinda in a bind with that one. Bad chips or using 32bit processors for 64bit software are the only things I can think of that makes sense because not everyone is having the same problems as we are, its only a handful of us but the processors should be reading the G-code regardless of the bit rate.  Speaking of which, are you using a computer that has a AMD or Intel processor? I'm using a AMD Ryzen Threadripper 1950x and Windows 10.  For some reason, Microsoft won't let me upgrade to Windows 11 with my processor and I haven't had a chance to upgrade yet. I'm not sure that would have anything to do with it since the Cura software seems to be working just fine on it but stranger things have happened. I guess it might be time to upgrade my computer again.

zeroku3, 2023-11-06:
Well, I use an Intel i7 8700k with Windows 11! totally different! I am also currently talking to anycubic, at least they respond and quickly, which is appreciated, and knowing that this happens to more people, I hope they find a solution quickly, and that with a software update only it can be fixed!

GregValiant, 2023-11-06:
I'll go ahead and close this as it appears to be an Anycubic problem.  You can still post here it will just be off the UltiMaker list.

closed 2023-11-06 by GregValiant

chfink0, 2023-11-23:
Sounds dumb, but please consider: Did any of you guys try if the power cord is attached **really** correctly? After some failing prints with a hung up kobra 2 max (always at the exact same spot) I came to the printer one day and it wouldnt turn on. After checking everything, I noticed that the main power cable had obviously minimal slack so I reattached it and secured it with cable ties. Since then I did not have issues. I know from other printers that they are very sensitive to power deviations, had the issue with a different printer before that it hung up after putting the lights off in the printer room. Also, with the massive speed and vibrations, may just be that at a certain movement, the connection comes just a little loose, not to much to restart the printer, but maybe enough to mess up the memory or whatsoever. Maybe you guys should try to fix the power cord and restart the faulty print...

rjack37, 2023-11-24:
No it wasn't the power cable.  The new updated firmware from Anycubic seems to have corrected the problem along with the suggested Cura profile settings, it hasn't locked up since. Now if I can only get a good print out of it without the PEI lifting of the magnetic bed and ruining my ASA prints, it might be a usable printer but for now I just keep wasting time and money with filament trying to get it to print correctly.

 
 

  

|  | Virus-free.www.avg.com |

rjack37, 2023-11-24:
The new Firmware for Anycubic Kobra 2 Max is:
Firmware Update Version 3.0.6 https://www.anycubic.com/pages/firmware-software or download it at  https://drive.google.com/file/d/1U3eobMOb2IgBkFM1IME7WMVEdEwIy_yI/view?usp=sharing 

Instructions: https://www.anycubic.com/pages/firmware-software or download them at https://drive.google.com/file/d/1DpMzThV0L3fP8EdZHv9vYpWJW4W0FpTt/view
Would you happen to know of a brand of ASA filament that doesn't warp so bad that it  pulls the PEI Spring Steel pulls off the magnetic bed with prints that run the entire length of the Print Bed?

 
 

new version?

  

|  | Virus-free.www.avg.com |

zeroku3, 2023-11-24:
it's official? It's not on the website! well let's try it!!! thank you. I only print PLA!ç

chfink0, 2023-11-24:
V3.0.6 is not for the MAX - 2.3.5 is the most recent for MAX.

rjack37, 2023-11-24:
Thank you!

 
 

  

|  | Virus-free.www.avg.com |

rjack37, 2023-11-24:
I was just told that the V3.0.6 is not for the MAX - 2.3.5 is the most recent for MAX.

 
 

  

|  | Virus-free.www.avg.com |

rjack37, 2023-11-24:
It shows on the anycubic support that the firmware for the Kobra2 series is V3.06 and there is not any firmware on the Kobra 2 max side, only firmware update instrctions.  Is it possible that the firmware is the same for the entire Kobra 2 series? 

 
 

  

|  | Virus-free.www.avg.com |

chfink0, 2023-11-24:
> Is it possible that the firmware is the same for the entire Kobra 2 series?

I doubt that because installation of this firmware is simply not possible on the MAX - I tried it on usb on all three ports - and also the general kobra 2 instructions for firmware are not applicable for the max, there is no TF card slot on the display for example.