Ultimaker/Cura #17035: PETG and PVA support slice preview fail every time

atlemo, 2023-10-19:
### Cura Version

5.4.0

### Operating System

macOS 14.0 (23A344)

### Printer

Snapmaker J1S

### Describe your model

Gyro_the_Dodo_by_Virtox

### Add your .zip here ⬇️

Printing with IDEX, using PETG and PVA, supports with 0 gap.
Fails every time I try to create G-code.

Cura File download: https://shared-assets.adobe.com/link/5818b57b-0cf4-4b28-601b-85db573934c3

GregValiant, 2023-10-20:
Thanks for the report and the file.
I wasn't able to slice this in either 5.4.0 or 5.5beta.
There is a bug in 5.4.0 that can keep a good model from slicing and I don't find any errors in that model.
There are three workarounds that alone or in combination can fool the bug.
Move the model in Cura
Rotate the model in Cura
Change the Initial Layer Height by a little bit.
I'll mark this as a duplicate and the Cura team will take a look.

atlemo, 2023-10-20:
Thanks for the quick response, I did try those suggestions as they were mentioned in the initial error dialog. It did work, but not very reliably.