Ultimaker/Cura #17042: Checked items are not showing up in the print settings, such as brim acceleration

wizardfirestore, 2023-10-20:
### Cura Version

5.5.0 beta. 1

### Operating System

MacOS Ventura 13.6

### Printer

Anycubic Kobra 2

### Reproduction steps

Checking the box under Settings Visibility, for brim acceleration.

### Actual results

When customizing the settings visibility under custom selection, checking the brim acceleration does not appear under Speed tab on the Print settings dropdown on the right of the screen.

### Expected results

Brim acceleration options should have appeared on the list of settings so that I can customize the settings.

### Add your .zip and screenshots here ⬇️

[image]
[CFFFP_test.3mf.zip](https://github.com/Ultimaker/Cura/files/13057124/CFFFP_test.3mf.zip)

GregValiant, 2023-10-20:
Thanks for the report.
This is from fdmprinter.def.json and is for "acceleration_skirt_brim":
"enabled": **"resolveOrValue('acceleration_enabled')** and (resolveOrValue('adhesion_type') == 'skirt' or resolveOrValue('adhesion_type') == 'brim' or resolveOrValue('draft_shield_enabled') or resolveOrValue('ooze_shield_enabled'))",
                    "settable_per_mesh": false,

In the project file you provided "Enable Acceleration Control" is not checked and so all the acceleration settings are hidden.  With Acceleration Control checked the setting is visible.  That's how it works.
[image]

I'll remove the bug label as this is a setting issue.  Let me know if we can close this.

GregValiant, 2026-09-10:
I'll close this.

closed 2026-09-10 by GregValiant