Ultimaker/Cura #17048: Crashes upon startup

jheinz2, 2023-10-20:
### Cura Version

5.4.0, 5.5.0 Beta

### Operating System

Windows 11

### Printer

Ultimaker S5

### Reproduction steps

Upon startup, software crashes. Every 3 - 5 times starting up it will finally load. Then after 2 or 3 changes within the software it crashes again. Things like changing material from the generic to the loaded Tough PLA, or loading a file will crash the software.  I've reloaded the software, upgraded to the Beta version, downgraded again. Issue happens in all scenarios. I've removed most plugins and materials that I'm not using from online account. 

### Actual results

Crashes either before startup, or after touching 3-5 settings or functions. 

### Expected results

Stay open. 

### Add your .zip and screenshots here ⬇️

[cura.log](https://github.com/Ultimaker/Cura/files/13058773/cura.log)

rmeissn, 2023-11-01:
I'm on Fedora 38, using the [Cura Flatpak](https://github.com/flathub/com.ultimaker.cura) from Flathub. Cura won't start any more and crashes while "Loading UI..." every time I'll start it.

dgarber1813, 2024-03-10:
I was having a similar issue on 5.4 so before I researched it too far I updated to 5.6 but I'm still having the same issue.... see attached pic
[image]

Can someone please tell me how to resolve???

jheinz2, 2024-03-11:
Sorry, I had other issues on my PC. Just had IT rebuild it. Everything works now, but don't know what would fix this. I wiped all settings from Cura, only thing I didn't try was a factory reset on the digital factory inside Ultimaker. Nothing I did seemed to fix it.