Ultimaker/Cura #17049: Manual entry for rotation angles

Slashee-the-Cow, 2023-10-21:
### Is your feature request related to a problem?

Sometimes when I'm trying to put something very precisely on my plate, sometimes because I'm reaching the edges of the plate and want to maximize my available area, sometimes I want to lay things at a certain angle to minimise support needed, or some models I have their default orientation isn't quite upright. It's hard for my to precisely (physical disability) rotate things using the rings and sometimes I want angles more precise than what can be achieved by snap rotation

### Describe the solution you'd like

Text boxes in the rotate flyout on the sidebar like there are already for the move and scale tools to set exact rotation angles

### Describe alternatives you've considered

About the only alternative I can think of is using another program to lay out my model/s and save that as a 3mf, but I haven't found a simple enough program that will do that while respecting the build area of my printer (other than maybe another slicer, but that would defeat the purpose)

### Affected users and/or printers

It benefits all printers. In my time on the forums I've seen several people looking for more precise ways of placing their model (in general) so while I don't know how much of the general user base will want it, it's not a "just me" thing. Probably a lot more people than CBF posting on the forums.

GregValiant, 2023-10-21:
@Slashee-the-Cow something you can do right now is to go into the "RotateTool.py" file ("C:\Program Files\UltiMaker Cura 5.5.0-beta.1\share\uranium\plugins\Tools\RotateTool") and you will see that line 49 is:
self._snap_angle = math.radians(15)

I have on occasion changed that to different angles and it works fine.  

You may be able to beat this request into a PR that utilizes an input box.

scotty6435, 2023-10-22:
Ideally, the UI should be updated to match the move and scale views. That way you can have granular controls over the actual rotation and you could even add a 'snap angle' input field to easily manage this. If this snap measurement is project-wide rather than per-object. this should sit within configuration

BMarik, 2024-05-03:
This is frustrating because earlier versions of Cura allowed the manual entry of rotation values. These newer versions do not. This feature needs to be reinstated.

fieldOfView, 2024-05-03:
>  earlier versions of Cura allowed the manual entry of rotation values. 

Not Ultimaker versions... I know Lulzbot had an implementation of entering rotation values in their Lulzbot Edition of Cura, perhaps you used that?

closed 2025-12-09 by GregValiant