Ultimaker/Cura #17051: Slicing Failed

Zestymato, 2023-10-21:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Ender 3 S1

### Describe your model

This model has not sliced for me at all. Every other file that came with it in a large paid for pack have all worked just fine. 

### Add your .zip here ⬇️

[CE3E3V2_makers-anvil-dwarf-settlement-stronghold-floor-c.zip](https://github.com/Ultimaker/Cura/files/13059930/CE3E3V2_makers-anvil-dwarf-settlement-stronghold-floor-c.zip)

Zestymato, 2023-10-21:
Sorry to bother you but right after posting this i managed to get it to work!
Ended up using the rotate tool and then using align to build plate and its happy now.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic