Ultimaker/Cura #17054: Slicing Problems

Vollblutasiate, 2023-10-21:
### Cura Version

5.5.0-beta.1

### Operating System

Windows 10

### Printer

Anycubic Kobra Neo

### Describe model location

The model slices if not rotatet, but does not slice when rotatet

### Describe your model

I sliced the model a few times before with Cura 5.3.1 without problems

### Add your .zip here ⬇️

[CFFFP_V2.zip](https://github.com/Ultimaker/Cura/files/13060519/CFFFP_V2.zip)

GregValiant, 2023-10-21:
Thanks for the report.
This model has some serious errors.  None of my tools was able to repair it.  I was able to slice it in 4.13.1.

The Cura team will take a look at this.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic