Ultimaker/Cura #17055: Sketchup skp file support

LightBridge16, 2023-10-21:
### Is your feature request related to a problem?

Sketchup support

### Describe the solution you'd like

hello,

I'm new toot the 3D printing hobby and drawing, but I have a Request for supporting skechtup file importing because I have a .skp file and I have no option too convert it for free.

would it be possible?

thanks for the latest beta version I love it!

### Describe alternatives you've considered

I can’t find a free alternative

### Affected users and/or printers

Cr-10S pro v2

scotty6435, 2023-10-21:
I've done some digging and you should be able to export a STL which can be sliced.

GregValiant, 2023-10-21:
Yeah, Sketchup is used by a lot of people.  [HERE ](https://help.sketchup.com/en/sketchup/importing-and-exporting-stl-files-3d-printing)are the instructions from the Sketchup website.

Always double check your SketchUp model to insure that all surfaces are facing the correct direction.  If surfaces are flipped inside-out (which seems to be easy to do) the models won't slice correctly (or at all).

I'll go ahead and close this as the functionality appears to be built in to SketchUp.  Let me know if you feel that is premature.

closed 2023-10-21 by GregValiant