Ultimaker/Cura #17056: Multiplied print failed to slice

AngelaCostello, 2023-10-21:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Ender 5

### Reproduction steps

I created a medallion in TinkerCad, and have no problems printing just one, but when I go to multiply, I get an "unexpected error."

### Actual results

It happens every time I try to slice a bed with multiple copies.

### Expected results

That there would be no issues with the slice.

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2023-10-21:
Thanks for the report but without a project file ("File | Save Project" and zip it) there is no way to tell what might be going on.

AngelaCostello, 2023-10-21:
> You don't often get email from ***@***.*** Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>

Hi Greg,

I was able to fix the issue by reducing the amount of multiplied objects to 7.

Best,
Angela

[EXTERNAL EMAIL] DO NOT CLICK links or attachments unless you recognize the sender and know the content is safe.

GregValiant, 2023-10-21:
If there is an error in a model, it might slice.  When you multiply the model you multiply the number of errors on the build plate.  Maybe it reached a point where Cura couldn't deal with the error count any more.

There is a bug in 5.4.0 that seems to jump up when models are in certain orientations.  Since your multiplied models are in a  haphazard orientation, I wonder if at least one ended up in 5.4.0's "I ain't gonna slice this" position?