Ultimaker/Cura #17057: Tried setting up Cura developent environemt

Toasterklient, 2023-10-21:
I hope it´s the right place to ask for help:

I want to join Cura developenet and looking more in to it. I followd the intructions from the Wiki Getting Started.

1. Downloaded PyCharm Community edition 2023.2.3
2. followed step 2
3. tried to go through step 3 of the instruction but it desent work. It said the compiler wasn´t definded, i checkt and modified the conan defaut profile. Definded a compiler and compiler version. After that the process started but didn´t complet.

What am i doing worng? Or should i try it in Visual Studio Code?

conan install . --build=missing --update -o cura:devtools=True  -g PyCharmRunEnv
Configuration (profile_host):
[settings]
arch=x86_64
arch_build=x86_64
build_type=Release
compiler=gcc
compiler.cppstd=17
compiler.libcxx=libstdc++11
compiler.version=13.1
os=Windows
os_build=Windows
curaengine:compiler.cppstd=20
curaengine*:compiler.cppstd=20
curaengine_plugin_infill_generate:compiler.cppstd=20
curaengine_plugin_gradual_flow:compiler.cppstd=20
curaengine_grpc_definitions:compiler.cppstd=20
scripta:compiler.cppstd=20
umspatial*:compiler.cppstd=20
dulcificum:compiler.cppstd=20
[options]
cura:devtools=True
[build_requires]
[env]
[conf]
tools.cmake.cmaketoolchain:generator=Ninja
tools.env.virtualenv:auto_use=True
tools.gnu:define_libcxx11_abi=True
tools.build:skip_test=False

Configuration (profile_build):
[settings]
arch=x86_64
arch_build=x86_64
build_type=Release
compiler=gcc
compiler.cppstd=17
compiler.libcxx=libstdc++11
compiler.version=13.1
os=Windows
os_build=Windows
[options]
[build_requires]
[env]
[conf]
tools.cmake.cmaketoolchain:generator=Ninja
tools.env.virtualenv:auto_use=True
tools.gnu:define_libcxx11_abi=True
tools.build:skip_test=False

WARN: grpc/1.50.1: requirement openssl/[>=1.1 <4] overridden by asio-grpc/2.6.0 to openssl/1.1.1l 
WARN: pyarcus/5.3.0: requirement zlib/1.2.12 overridden by cura/5.6.0-alpha to zlib/1.2.13 
WARN: cpython/3.10.4: requirement zlib/1.2.11 overridden by pyarcus/5.3.0 to zlib/1.2.13 
WARN: tcl/8.6.10: requirement zlib/1.2.12 overridden by tk/8.6.10 to zlib/1.2.13 
WARN: curaengine/5.6.0-alpha+testing_1aedf1@ultimaker/testing: requirement zlib/1.2.12 overridden by cura/5.6.0-alpha to zlib/1.2.13 
WARN: spdlog/1.10.0: requirement fmt/8.1.1 overridden by curaengine/5.6.0-alpha+testing_1aedf1@ultimaker/testing to fmt/9.0.0 
WARN: curaengine_plugin_gradual_flow/0.2.0-alpha+testing_df1e5e@ultimaker/testing: requirement curaengine_grpc_definitions/0.1.0-beta.1 overridden by cura/5.6.0-alpha to curaengine_grpc_definitions/0.1.0-alpha+testing_c50918@ult
imaker/testing
gnu-config/cci.20201022: Not found in local cache, looking in remotes...
gnu-config/cci.20201022: Trying with 'cura'...
[... 138 lines cut ...]
    gnu-config/cci.20201022 from 'cura' - Downloaded
    gtest/1.12.1 from 'cura' - Cache
    msys2/cci.latest from 'conan-center' - Cache
    nasm/2.15.05 from 'cura' - Downloaded
    protobuf/3.21.9 from 'cura' - Cache
    standardprojectsettings/0.1.1@ultimaker/stable from 'cura' - Cache
    zlib/1.2.13 from 'cura' - Cache
Build requirements packages
    catch2/3.4.0:aa9e6cbf683ff9f2f8e3edf5ee73a574a7ce68db - Build
    cmake/3.26.3:ca33edce272a279b24f87dc0d4cf5bbdcffbc187 - Download
    gnu-config/cci.20201022:5ab84d6acfe1f23c4fae0ab88f26e3a396351ac9 - Download
    gtest/1.12.1:52ed1ad4f166351da9b32c79a3349f1f4d015057 - Build
    msys2/cci.latest:eee3fba89db6d777329de604625af8c30d46f080 - Download
    nasm/2.15.05:0a420ff5c47119e668867cdb51baff0eca1fdb68 - Download
    protobuf/3.21.9:3989277b79b140eb22d88222d97298fb1e887a16 - Build
    standardprojectsettings/0.1.1@ultimaker/stable:5ab84d6acfe1f23c4fae0ab88f26e3a396351ac9 - Download
    zlib/1.2.13:a86b08dfda6e449c9658163512714d447bd0edff - Build

Installing (downloading, building) binaries...
ERROR: There are invalid packages (packages that cannot exist for this configuration):
gdbm/1.19: Invalid ID: gdbm is not supported on Windows
cpython/3.10.4: Invalid ID: Invalid transitive dependencies
pyarcus/5.3.0: Invalid ID: Invalid transitive dependencies
pynest2d/5.3.0: Invalid ID: Invalid transitive dependencies
pysavitar/5.3.0: Invalid ID: Invalid transitive dependencies

Zincr0, 2024-01-08:
Note that current build scripts also fail on linux. It tries to install `protobuf==3.21.9` which is not a valid version.

jellespijker, 2024-01-09:
> Note that current build scripts also fail on linux. It tries to install protobuf==3.21.9 which is not a valid version.

@Zincr0 This probably because we're currently working on our Artifactory (which stores the conan packages). It should no work again for third-parties and community contributions. Please try the conan install command again.

lbibass, 2024-01-14:
> @Zincr0 This probably because we're currently working on our Artifactory (which stores the conan packages). It should no work again for third-parties and community contributions. Please try the [...]

It still attempts to install protobuf 3.21.9 on MacOS, which does not exist.

jellespijker, 2024-01-14:
Is your conan config up to date?
```
conan config install https://github.com/Ultimaker/conan-config.git
```