Ultimaker/Cura #17058: [5.5.0b1] "Print as support" model is ignored or actively avoided by support structure

Bzzz, 2023-10-21:
### Cura Version

5.5.0b1

### Operating System

Ubuntu 22.04

### Printer

Ender 5 Plus

### Reproduction steps

When working on a more complex model, I came to the conclusion that some custom support structure would be beneficial to reduce overall support material usage and print times, as Cura makes wild structures that can weigh up to 4x the actual model. The custom support model however is ignored or actively avoided by Cura when calculating supports.
1. Make model that requires support to print
2. Make a rough sketch for that support structure
3. Import both to Cura and set the support to "Print as support" (Check that "Drop down mesh support" isn't active, although it doesn't really make a difference here - https://github.com/Ultimaker/Cura/issues/12653)
4. Slice

### Actual results

5. Watch Cura ignore or avoid your custom support and do its own thing. 

If the model is printed (e.g. when disabling additional supports, but not limited to that option), parts smaller than 2x nozzle diameter (single-walled) are ignored entirely

### Expected results

5. See that:
5.1 your support is printed as support, and 
5.2 additionally needed support (e.g. the interface layers between your support model and the actual print) are added, spanning from the most convenient spot on the support model to the print. No parallel structures from build plate to support layer are expected

### Add your .zip and screenshots here ⬇️

[image]
[image]
[image]
[image]
[image]

I've created a dumbed-down model for easier/faster slicing, featuring similar overhangs and support structure. The support ring cutout is intentionally placed off-center to illustrate effects on wall thickness <2x nozzle diameter (0.6mm nozzle size).
In the full model regular supports double up the custom support with their own structure instead of using it. Tree supports grow alongside the support structure instead of growing from the support where needed. They also sprout from the model when plenty of free space nearby is available; suppressing this by using the "buildplate only" option creates its own issues when supports on top of the model are advantageous.

Print settings don't really affect this issue much, I get pretty much the same issue with the Cura base settings in standard or draft mode (with support enabled, of course). 

[curabug.zip](https://github.com/Ultimaker/Cura/files/13061092/curabug.zip)

GregValiant, 2023-10-21:
Thanks for the report.
Sorry, but I don't think I understand this.
On the left is your model and I've brought in a cylinder with a concentric hole on the inside.  It is configured to "Print as Support", "Generate Support" is turned on, and I've added a support blocker to keep the Cura generated supports from being added.
On the right is your project.
[image]

Bzzz, 2023-10-21:
What do you need the support blocker for? If I had a perfectly matching support model, I'd just generate without support entirely. Thing is: I don't, I don't know the proper spacing for the easy surfaces, and I have no clue how to tackle everything non-rectangular. So I do want support, just with my cylinder as the base of all added supports.

GregValiant, 2023-10-22:
How about if you set that block to print as "Normal Model" with single wall, no top, no bottom and no infill?

I don't think Cura has the capability to provide "Support for Support".  You could put in a feature request, but I don't see a bug here.

Bzzz, 2023-10-22:
There's another bug that doesn't let "print as support" models switch back to normal models, but that aside your suggestion works for the demonstrator:
[image]
So the model is printed underneath, gets some interface layers and then supports the main model. That's what I would expect for the "print as support" option in the first place (maybe we should rename the thing if it has a different purpose? Can you explain what it is actually good for?)

For the full thing I'm trying to print, it does not work. There's support around the support model when using the default Cura profiles and total weight of the print is 212g with the base model clocking in at 44g. I did double the wall thickness yesterday plus moved it closer to the model, but haven't checked what happens when I go back. Given the top is still considered too narrow despite matching the ring it's supposed to support, I'm having some doubts.
[image]
With my custom profiles I'm getting anything from slicing failures to weird supports ending mid-air without supporting anything, to double structures again.
[image]
[image]