Ultimaker/Cura #17059: Tapa Litofanía 16cm

LasmanosdeLuna, 2023-10-21:
### Cura Version

5.4

### Operating System

Windows 10

### Printer

Artillery genius

### Name abnormal settings

No, No

### Describe model location

No

### Describe your model

Siempre. Sí, lo he hecho pero nada

### Add your .zip here ⬇️

[ezyZip.zip](https://github.com/Ultimaker/Cura/files/13061516/ezyZip.zip)

GregValiant, 2023-10-22:
Thanks for the report.
The model doesn't appear to have any errors.  I can't get it to slice in 5.4.0 or 5.5beta but it did slice in 5.2.2 and 4.13.1.

The Cura team will take a look.
This is in 5.2.2
[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic