Ultimaker/Cura #17062: 3MF Thumbnail and QuickLook support in macOS

fluxor99232, 2023-10-22:
### Is your feature request related to a problem?

As opposed to STL files I don't get thumbnails or QuickLook previews of 3MF files in macOS (13.6 Ventura) Finder with Cura 5.4. Cura is my default file handler for these files. In my understanding this is something that would have to be implemented by Cura.

### Describe the solution you'd like

Get thumbnails and QuickLook functionality in macOS Finder for 3MF files.

### Describe alternatives you've considered

I did a quick Google and Cura Marketplace research and couldn't find any apps or plugins to achieve this.

### Affected users and/or printers

macOS users

### Additional information & file uploads

[image]
[image]
[image]

craigrusso, 2024-10-23:
this would be great to have.

fluxor99232, 2024-10-29:
Nevermind, I've switched to OrcaSlicer...