Ultimaker/Cura #17063: Ultimaker S3: Prime blob location in front of camera

ogsadmin, 2023-10-22:
### Cura Version

5.5.0-beta.1

### Operating System

Windows 10

### Printer

Ultimaker S3

### Reproduction steps

Slice and run a job with prime blob enabled. Then monitor the print progess on the monitor tab through the Ultimaker S3 integrated camera.

### Actual results

The prime blob is put into the front-right corener - exactly where the camera is located, so the prime blob obscures the camera view.
Here is a sample screenshot:
[image]

### Expected results

Put the prime blob by default into any other corner (front left maybe or back right).

In general, also the hardware location/angle of the camera seems to be non-optimal - it mostly shows a side view of the plate instead of the printed object.

### Add your .zip and screenshots here ⬇️

Here is the screenshot (same as added to the "Expected Results" section):
[image]

GregValiant, 2023-10-22:
Thanks for the report.
I'm not an UltiMaker guy so @MariMakes and the Cura team will take a look and correct this if I am wrong.

There is a move prior to the prime blob (G280 line) that you could change using "Search and Replace".
When I generate a gcode there is this near the start:
;Generated with Cura_SteamEngine 5.5.0-beta.1
T0
M82 ;absolute extrusion mode
G92 E0
M190 S60
M109 S205
**G0 F9000 X-3 Y6 Z2
G280**
The G280 instructs the printer to perform it's "purge blob" routine at the location that it just moved to.  If you add another move prior to the G280 you could put the prime blob wherever you want on the build plate.

Search and Replace
Search = G280
Replace = G0 X230 Y190\nG280
Regular Expression = not checked

that would result in this
G0 F9000 X-3 Y6 Z2
G0 X230 Y190
G280

So the added move (to X230 Y190 in that example) can be to wherever you want on the build plate.  The "backslash+n" in the replace line is a newline character and is required.  (If parameters are missing from a line the printer will maintain the settings from a previous line and so "F" and "Z" parameters aren't necessary in the "replace" box but you could add them if you want.

ogsadmin, 2023-10-22:
Hi @GregValiant, thanks for your detailed explanation and how to work around this.
 
However, (at least for me) the major point in buying an Ultimaker (S3) printer is the seamless and reliable workflow it allows: run Cura, import an STL file, click slice and send to printer - that's it. No need to fiddle around with GCodes...

So even though this seems to be a minor point (and I agree, it actually is), I feel like this should be fixed (and I am sure it is a simple fix, it's just the default coordinate for the prime blob). It's Ultimaker software and Ultimaker hardware - and it should work perfectly together :-).

Thanks and keep up the great work!

MariMakes, 2023-10-23:
Hey @ogsadmin,

Thanks for your report 👍

It's been a while since I've seen this request, we used to see it way more often. But during the launch of the Ultimaker S5, in april 2018, we learned that our Sline printers don't need a prime blob, and that printing the skirt is enough to prime the extruder. So prime-blobs are disabled by default, and our users have to turn them on actively. 

Can you share why you need the primeblob for this printjob, instead of the skirt? 

I also see that you have your active extruder on the right, and the support extruder on the left, which I see happening even less than people enabling the primeblob. If you turn the printcores and materials around your primeblob will not be placed in front of the camera but on a different corner. Did you choose this extruder with a specific intention?

ogsadmin, 2023-10-23:
Hi @MariMakes, that now makes sense! 

As I've ruined a few glass plates already (due to ripping of parts of the glass surface when removing a part - even with using the gluestick), I've switched to using a BuildTak PEI plate. I am pretty happy with it, except, that the skirt sticks that badly, that one almost cannot remove it. So I switched to the prime blob (for mostly flat prints, that don't need additional bed adhesion) to get the nozzle/flow initialized before the print starts.

Thanks for the hint with the extruders - I did actually not really think about this. Maybe it was an oversight at the time I installed them - and I've got used to working this way...  

I'll swap the extruders - then this will fix it!

Thanks for your support!

MariMakes, 2023-10-24:
Great to hear 😊
Since there is a workaround in place, I'll be closing this issue. 

Thanks again, and please let us know if you run into any other issues 💪

closed 2023-10-24 by MariMakes