Ultimaker/Cura #17068: Add option to 'ignore' printer settings on project open

scotty6435, 2023-10-22:
### Is your feature request related to a problem?

When printing a multi-project setup, often along the way the printer setting change. Integrating these changes into the next project as when you open the project with the updated profile already loaded, you are only give the option to:

- Update (overwrite the current settings)
- Create new (create a new profile)

You have to 'create new' to avoid losing the old profile which just doesn't need to happen.

### Describe the solution you'd like

Add a third option; "Do not import" or similar. Doing so will load the project with the currently active printer profile.

### Describe alternatives you've considered

Other than noting the changed settings in another application or intentionally re-saving the amended profile, I'm not sure what else could be done

### Affected users and/or printers

All

Asterchades, 2023-10-23:
There's already the option to only import the models from a project file, rather than the entire project. This does what it sounds like it should - absolutely zero settings are imported, no new profiles are requested, and if you're missing the specific printer involved it won't be added to your configuration.

It is, however, possible to skip being asked the question as to which you want to import. If you've already skipped this in the past you can restore it by opening the configuration (Preferences -> Configure Cura), scrolling down to the bottom and adjusting the `Default behaviour when opening a project file` drop-down (immediately _above_ the Profiles header).

scotty6435, 2023-10-25:
Thanks @Asterchades, I didn't realise that you could open a project and just import the files, I'll do that in future.

closed 2023-10-25 by scotty6435