Ultimaker/Cura #17069: unable to slice, even with 0.2 layer hight

pcmueller77, 2023-10-22:
### Cura Version

5.4 and 5.5

### Operating System

Windows 11

### Printer

Sovol SV06 +

### Reproduction steps

I added the profile and the stl. Even if reducing the layer high from 0.28 to 0.2 its not slicing

### Actual results

slicer stopped calculating

### Expected results

to be able to slice or see a hint regarding a bad setting

### Add your .zip and screenshots here ⬇️

[SpoolHolderGuide_complete_bearing V2.zip](https://github.com/Ultimaker/Cura/files/13063167/SpoolHolderGuide_complete_bearing.V2.zip)

GregValiant, 2023-10-22:
Thanks for the report and the file.
My impression is that Cura and Windows 11 systems appear to have continuing issues when slicing.  I was able to slice your project on Windows 10 with no problem so I can't duplicate this behavior.
Then there is this model...
Cura is informing me that the model has errors.  MS 3D Builder says no errors.  The on-line repair tool I use (https://formware.co/OnlineStlRepair) says no errors and then timed out trying to repair the model.  Netfabb hung trying to repair this model.  My experience says there are subtle errors in the model and they are apparent in the Cura x-ray view as rectangles within the model.  The three top areas I've highlighted appear to be bad internal surfaces.  The lower larger area might be a design intent, but may be do to mis-assembly of the model in CAD.
[image]

This is how that larger low area sliced.  Is this how the model is supposed to look?  You can see there is a gap there and it's definitely in the model.
[image]

If you could provide more information regarding the computer system you are using (manufacturer, motherboard, CPU, RAM, etc.) it might be helpful.

pcmueller77, 2023-10-22:
Hello Greg Valiant, thanks for the very fast answer. 
I was not able to slide it with the profile, attached. I am not able to set the layerhigh higher than 0.2 for a faster print.
The gaps are not wanted :-) its a spoolholder for the side for an sovol sv06. the arm for  spool is not in this model, its an extra project with bearings and a metal bar, which is connected to the hole in the middle of the project her. I just ask, if it would be helpful, if cura tells me mismatched settings.. but i guess there are none, its the model as  you mentioned. Okay I made it with Microsoft 3d builder. Is there a tool to fix errors? Thanks a lot for your help, I will fix the model & best regards, Thomas.

GregValiant, 2023-10-22:
The model needs work, but I didn't have any trouble using the Standard profile for your printer and at .28 layer height.
You can't print this fast, it's too tall and will shake and make for a sloppy print even if it doesn't fail.
Here is another place with poor assembly.  I'm guessing that round feature at the top should be square to the world.
[image]

GregValiant, 2023-10-23:
@pcmueller77 I'm inclined to remove the bug label and go ahead and close this.  It certainly appears to be a model problem rather than something going on with Cura.  Let me know if you would agree with that.

pcmueller77, 2023-10-27:
okay, I still can't print with the cura profile I attached, but the model had issues. So with a standard profile I can slice, but not with the one I attached in this issue. So: Is there a way to find a mismatch setting in the profile?
Thanks & with best regards, Thomas.