Ultimaker/Cura #17077: Initial layer speed

Geoschatter, 2023-10-23:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Elegoo Neptune 4 pro

### Reproduction steps

Filament : E-sun PETG
If I set the initial layer speed to 105.0 mm/sec. the effective speed is 5.0 mm/sec.
If I set the initial layer speed to 160.0 mm/sec. the effective speed is 60.0 mm/sec.

### Actual results

Apparently the slicer does not see the first 100 mm/sec. 
[image]
[image]

### Expected results

The final print speed should be equal to the Initial layer speed as set in the slicer, shouldn't it ? 

### Add your .zip and screenshots here ⬇️

https://github.com/Ultimaker/Cura/assets/56678150/d52ae98b-5d5d-4a82-b9e5-5af9d26ce36e

https://github.com/Ultimaker/Cura/assets/56678150/d52ae98b-5d5d-4a82-b9e5-5af9d26ce36e

GregValiant, 2023-10-23:
Thanks for the report.
I am unable to duplicate this behavior.  The actual speed is subject to other settings like "Minimum Layer Time" and the number of slower layers.
If you would provide a project file ("File | Save Project" and zip it) it would proper information to investigate this.

Geoschatter, 2023-10-24:
Printer : Elegoo Neptune 4 pro.
Filament : E sun PETG.

Attached is the project file as You requested, hopefully you can do
something with it because I don't know much about it myself.
In the other attachment "Error.png" you can see in the upper part the error
message when I set the Initial Layer Speed lower than 100 mm/sec. if I set
it above 100 mm/sec. there is no error message.
During printing itself the print speed is 100 mm/sec. slower than set in
the slicer.

With kind regards ,
Erwin Lemmens

GregValiant, 2023-10-24:
Al attachments must be made here on the GitHub website.  Emails attachments don't come through.

MariMakes, 2023-11-21:
Hi @Geoschatter,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Could your issue be related to: https://github.com/Ultimaker/Cura/issues/14874?
We see that there are a number of settings that are "competing" to define your initial layer speed.

Sometimes that results in Top Bottom Flow, Initial Layer Flow and Initial Layer Bottom Flow all competing to define the setting. 
You can usually see an indication of the print speed in the preview. Here you see in indication of the flow if you change the settings. 
[image]

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

closed 2023-12-05 by github-actions[bot]