Ultimaker/Cura #17086: Link

Swatton-A, 2023-10-24:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Ender 3 v2

### Describe your model

Fix would not work, and has sliced successfully previously

### Add your .zip here ⬇️

[CE3E3V2_Link_Skyward_Sword v2.zip](https://github.com/Ultimaker/Cura/files/13111812/CE3E3V2_Link_Skyward_Sword.v2.zip)

GregValiant, 2023-10-24:
Thanks for the report.
That model has a lot of errors.  A second problem is that something is wrong with that "Cura5-4mini.." setting profile.  After repairing the model and switching to the Super Quality profile it sliced with no problem
Here is your model on the right and the repaired model on the left.  The base is particularly bad and is non-existent around the periphery.
[image]

I have no idea what might be wrong with the Cur5-4..." profile.  Here is the repaired model with the Super Quality profile.
[image]

Between the model problems and the bad profile there is no way to tell if this might be a bug.  I'll remove the bug label and go ahead and close this.  I hope you understand.

closed 2023-10-24 by GregValiant

Swatton-A, 2023-10-25:
Many thanks for this, I am very new to the 3D printing scene and am learning a lot daily.  Could I asked what the process was that you followed to repair the model?

GregValiant, 2023-10-25:
Since you are using "Windows" you can go to the Microsoft Store and download MS 3D Builder.  It's repair utility is pretty good and it gives you the ability to alter STL files.
There are a number of on-line sites that will come up with a search.  I like [formware.co](https://formware.co/OnlineStlRepair) because it gives a report on the types of errors that are present in a model.  It has a size limit for the files and a time limit when attempting a repair so it's a bit limited.
Don't assume that because you are having trouble that there must be a bug in the software.  There are bugs in the software, but not a lot.  What Cura is not real good at is telling you _why_ something is going on and confusing it.  It is written for Linux, MAC, Windows and a couple of other operating systems.  That commonality makes it easier to maintain, but since it isn't native to any particular OpSys it is clumsy.  A case in point would be that profile.  I can't tell what's wrong with it, but something is in conflict because it doesn't work.  Start with the included profiles for your Ender (Super, Standard, etc.) and make a couple of changes at a time.  Almost every model will need the settings tweaked at least a little bit.