Ultimaker/Cura #17093: Failed to slice

smaj100, 2023-10-24:
### Cura Version

5.4.0

### Operating System

windows 10 home

### Printer

Ender 3 v2

### Describe your model

yes it sliced previsouly with parts in a different location on build plate

### Add your .zip here ⬇️

[planetary gearbox.zip](https://github.com/Ultimaker/Cura/files/13115178/planetary.gearbox.zip)

GregValiant, 2023-10-26:
Thanks for the report and the file.
5.4.0 has a bug that can keep models from slicing.  It doesn't appear to be completely fixed in 5.5.0 but this project did slice for me in 5.5.
You can try moving the models around on the build plate.  That often works to fool the bug along with changing the Initial Layer Height by a little bit.

I'll mark this as a duplicate and go ahead and close it as the new version is out now and it did slice in it.  I hope you understand.

This is in 5.5.0
[image]

smaj100, 2023-10-26:
Greg,  thanks i did move the parts around and sliced it successfully.
Lee

[image]<https://user-images.githubusercontent.com/64202104/278182058-e0db6f5c-1229-4e0b-8e29-31b9f06ad265.png>