Ultimaker/Cura #17094: [5.5] beta (and every version I've tried including 4.11 and 5.4), incorrectly slices with one too-large layer near the bottom

FlakManiak, 2023-10-24:
### Cura Version

Version 4.11, version 5.4.0, and version 5.5.0 beta 1

### Operating System

Windows 10, NVIDIA GeForce RTX 2070

### Printer

Creality Ender 3/Ender 3 V2

### Reproduction steps

I should begin by saying that I have had this issue with, as far as I can tell, every single STL I have sliced in Cura, with every Cura version I have installed, even across fresh re-installs. (Those Cura versions being 4.11, 5.4.0, and 5.5.0 beta 1.)

The example I will use now is this attached cube STL (included in the zip file), made in Blender. It's a plain cube.

Cura is set to use the Creality Ender 3/Ender 3 V2.

I load the STL in Cura, then I set its size to 1000%. This results in a 2cm cube. I choose "Super Quality" (though this effect is visible on other stock quality settings, it's more-visible here). All the Super Quality settings are default; I know this because I did a clean install, even deleting the AppData folders, beforehand. (I did this clean-install process on 5.4.0 and on 5.5.0 beta 1.) Specifically, I'll note that the "Horizontal Expansion" and "Initial Layer Horizontal Expansion" settings are at 0.0mm, their default values.

I hit slice, and then show the preview. At this point, Cura displays what looks like a properly-sliced cube. This, however, is a lie. I save the Gcode to disk. Then I load that Gcode file I've just made. This Gcode SHOULD now be a plain cube, with every layer the same size.

### Actual results

When this Gcode file is loaded, the bottom N (in this case two) layers are slightly smaller than most of the cube. Then right above them, exactly one layer is significantly too large (in this case layer three). The number of layers before the wrong-layer depends on layer height. (For example, with layer height set to 0.05, the too-large layer is layer 5.) The wrong layer is significantly too wide, and the layers below it are slightly too narrow. I've used a 2cm cube as my example because it's the simplest, but this has happened to every single thing I have sliced with Cura, be it old Gcode I made in 2021 on Cura 4.11, or new Gcode I made recently on 4.11, 5.4.0, and 5.5.0 beta 1.

I know that this is an actual problem in the Gcode and not in Cura's display of the Gcode, because that too-large layer does appear as a line on the physical prints. It looks like a mold line, with additional wispy plastic that didn't adhere because that layer was too wide to be printed properly.

I also know that this isn't a problem with my STLs, for the following reason: The STLs are generated in various orientations in Blender, and some are rotated and some aren't after being put into Cura. But this too-wide layer always appears near the bottom of the Gcode for every STL I slice.

Additionally, a friend of mine was able to reproduce this issue, but strangely, only on one version of Cura, 5.4.0. He had a different printer set, an Anet A8. When slicing this cube on 4.12.1, it sliced correctly for him. On version 5.4.0, he saw the same wrong-layers that I got. Then, when he switched to version 5.5.0 beta 1, it worked correctly for him again. This stands in contrast to how it works incorrectly for me on every version I have tried. But I think it's relevant that he saw this issue even when selecting a different printer-model; it's not solely due to Creality Ender 3 default values in Cura.

I have attached five screenshots. The first and second show the faulty Gcode, and you can clearly see the too-wide layer. The third screenshot shows the first two layers of the print. The fourth screenshot shows the third (too-large) layer. The fifth screenshot shows four layers of the print, and you can see that the fourth layer is significantly smaller than the third.

[image]
[image]
[image]
[image]
[image]

### Expected results

Cura should slice a cube into layers that are all of the same size. It should not do... This.

### Add your .zip and screenshots here ⬇️

I have provided a zip file with my project file, the Cura log file, and also the cube STL, and the screenshots of the faulty Gcode displayed in Cura, and also, much to your chagrin, various Gcode files I have recently generated that all show this same issue. I hope that you can figure out what's going on here, because I am well and truly baffled. Thank you for your time.

[CuraDebugStuff.zip](https://github.com/Ultimaker/Cura/files/13121093/CuraDebugStuff.zip)

GregValiant, 2023-10-25:
Thanks for the report.
First off understand that I don't work for UltiMaker and never have.  I'm just an old retired guy who does this as a hobby.  I have been working with gcode for over 50 years.  Consider that more than 100,000 people use the software.  If it couldn't slice a cube correctly nobody would use it.
I have written a couple of macros to read gcode files into AutoCad.  I have a couple of other macros that I use to analyze gcode files with MS Excel.  I have never seen what you are describing and I don't see it in the gcode files you provided.

This is "Poor slicing Cube Final.gcode" read into AutoCad.
[image]

This is the same file in orthographic view with the skirt turned off.  This looks as it should and as you would expect.  The side walls of this cube are exactly the same to 8 decimal places.
[image]

Here is the same file in a Top View.  All 100 layers are here and yet it looks like there is only one.  That's because they are exactly right above one another.
[image]

This is one of the BadGcode files.  Again, I don't see anything odd here.
[image]

This is one with a .12 Initial Layer Height, .05 Layer Height, and what appears to be 0.1 Initial Layer Horizontal Expansion but it isn't.  The line is offset by .1mm because this file was sliced with the "Initial Layer Line Width" at 150% making each line 0.6 wide with the expected result that Cura moved the path over by 0.1mm to account for the added line width.
[image]

These snippets are from the gcode file "2023PoorSlicingCubeNewCuraB"
from LAYER:0
;TYPE:WALL-OUTER
G1 F300 X127.2 Y107.8 E18.12116
G1 X107.8 Y107.8 E18.70188
G1 X107.8 Y127.2 E19.2826
G1 X127.2 Y127.2 E19.86332
from LAYER:1
;TYPE:WALL-OUTER
G1 F900 X127.3 Y107.7 E36.00152
G1 X107.7 Y107.7 E36.1645
G1 X107.7 Y127.3 E36.32747
G1 X127.3 Y127.3 E36.49045
You can see that they are different.  That's because of the "Initial Layer Line With" at 150%.
Here are the bottom few layers of the gcode in Cura preview.

[image]

Here they are when "Force Layer View Compatibility Mode" is enabled in Cura.
[image]

I don't see a bug here and I don't see the abhorrent gcode you describe in your report.  I'll remove the bug label from this but I'll leave it open so you can respond.

FlakManiak, 2023-10-25:
Please don't remove the bug label. Even if the Gcode were fine (and I don't believe it is, as this too-wide layer-line shows up in the physical prints), it would still be a bug in Cura's Gcode viewer, as seen in the sliced-view screenshots in my initial post. There's a bug here either way, even if it were a different type of bug.

I can't explain why several other Gcode viewers (including PrusaSlicer's) display this cube Gcode as being a straight-sided cube. But it doesn't print that way! If this were merely something wrong showing up on my physical prints, I would assume that there was some quirk with my physical printer. But if that were the case, why can Cura accurately predict this too-wide layer? Why does Cura know this will happen, when displaying Gcode? If Cura's Gcode really is correct, then why does the printer handle it wrong in the real world, and why does Cura's Gcode-viewer just _happen_ to display something that matches what the printer will do?

FlakManiak, 2023-10-25:
Also, in your Cura Gcode viewer screenshot, you can still see the behavior I'm talking about. Even if the very-first layer's dimensions are explained by the "Initial Line Width" setting, you can see that the second layer in that screenshot is both too wide and very thin. It's clearly thinner than the layers above and below it, and wider too.

smartavionics, 2023-10-25:
Hi, I sliced your project using my Cura variant (https://github.com/smartavionics/Cura/releases) and the layer display looks fine. I also looked at the gcodes you supplied with another gcode viewer and they looked fine.
[image]

GregValiant, 2023-10-25:
My printer is an Ender 3 Pro.  I have printed a lot of calibration cubes and I print a lot of mechanical models that start with rectangular shapes on the bed.  I have never had a problem like you are describing.  As I pointed out, if this was a problem with Cura then at least one of the hundreds of thousands of people using Cura (including myself) would have noticed and complained.  Over the past couple of years I think I might be the leader in the number of bug reports filed.

Somehow, someway, I think this has to do with your printer.  When I received mine from Creality several sub-assemblies that they had put together were highly out of adjustment and needed attention.  One fix was to shim the Z motor bracket by .015" to eliminate binding.  It was way off.
The problem might be something with the Z system, or bed leveling, or something else, but the problem does not appear in the gcode generated by Cura.  I've never seen it in the thousands of slices I've done and analyzed, and I don't see it in yours.  @smartavionics is a contributor to Cura and he has taken a look at the files you submitted and he concurs.

It could be that the preview is showing odd things.  Cura works by layers so for something like running the Infill Layer Thickness at 2X layer height - Cura will preview a line as double-width rather than double-height because that's the way it works.  It can't go down and put half the plastic on a previous layer, but the flow is double so it does what it can to show that.

If the point of all this is to fix your prints then I think you need to look at the machine again.  I bought my printer because it was cheap.  It comes with cheap fans (all replaced) a cheap hot end (replaced) cheap trolley wheel bearings (all replaced), etc.  Right now it has been sitting overnight unused.  The Hot End thermistor reads 27° and the bed thermistor reads 24°.  They are both sitting at room temperature.

GregValiant, 2026-09-10:
I'll go ahead and close this.

closed 2026-09-10 by GregValiant