Ultimaker/Cura #17096: slicing failed

rams78, 2023-10-24:
### Cura Version

5.4.0

### Operating System

win 12

### Printer

flying bear ghost 5

### Describe your model

no

### Add your .zip here ⬇️

[bottom_part.zip](https://github.com/Ultimaker/Cura/files/13122089/bottom_part.zip)

GregValiant, 2023-10-26:
Thanks for the report.
This bug seems to have carried over from 5.4 into 5.5.  I moved the model and rotated it 180°, and changed the layer height to .12 and it sliced.

I'll mark this as a duplciate and the Cura team will take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic