Ultimaker/Cura #17099: Inconsistent infill in layers of thin cylinders.

Flopalop2, 2023-10-24:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Voron 2.4 350

### Reproduction steps

1. Add cylinder stl to cura that is roughly 4 mm in diameter (there's a range like 3-6 but I know 4 works) and fairly tall. 
2. Slice with 2 walls and 20% infill of type cubic, cubic subdivision, octet, quarter cubic, and partially gyroid as well.  

### Actual results

[image]
[image]
[image]
Infill in sections of layers but inconsistent throughout part.

### Expected results

[image]
Expected infil throughout or not at all.

### Add your .zip and screenshots here ⬇️

[tubetest.zip](https://github.com/Ultimaker/Cura/files/13138098/tubetest.zip)

GregValiant, 2023-10-25:
Thanks for the report.
That is happening because you are using Gyroid infill on a small part.  You have to think of infill as filling your entire build plate and Cura uses the parts of that "giant cube of infill" that intersects your model.  Because Gyroid wanders around; for some layers there is no intersection with your small model.
You can kind of see what is going on by making the "Infill Density" 5% and then 75% and running the preview slider up and down.
Another test would be to scale the part in the X and Y to 25mm with Gyroid at 20% density.
Gyroid is kind of cool to watch print but it's only good for supporting the roof of a model and adds about zero strength to a print.

To print that piece with consistent infill switch to Grid.

I'll go ahead and remove the bug label and close this as it is expected behavior from Gyroid, Octet, Quarter Cubic, and a couple of others.  I hope you understand.

closed 2023-10-25 by GregValiant

Flopalop2, 2023-10-25:
Alright that makes sense thanks.