Ultimaker/Cura #17100: I can't change the build plate temperature

brunosartori08, 2023-10-25:
### Cura Version

5.5.0

### Operating System

Windows 10

### Printer

Faber S

### Reproduction steps

The marker is marked in the option to see "Build Plate Temperature" but when I will change it, it does not appear

### Actual results

The option "Build Plate Temperature" does not appear 

### Expected results

The field to change the "Build Plate Temperature" appears

### Add your .zip and screenshots here ⬇️

https://github.com/Ultimaker/Cura/assets/124313601/e884b58b-dbff-43a0-8b55-4bf4cfbccfcd

GregValiant, 2023-10-25:
Thanks for the report.
Where you tried to make the setting "visible" there is a "circle i" icon there.  Hover the mouse over it and see what the popup says.
If your printer is not configured with a heated bed then the setting will remain hidden.  Go to Manage Printers and then Machine Settings and see if the "Heated Bed" box is checked.
Let us know what you find.  If the setting remains hidden and the Heated Bed box is checked then use the "File Save Project" command, zip the project file, and post the zip folder here.

brunosartori08, 2023-10-25:
thanks you so much, it worked xD

closed 2023-10-26 by GregValiant