Ultimaker/Cura #17103: Slicing error when using higher resolution print, OK for lowest resolution.  The submission form does not allow me to attach the .3mf filettach the

Yorkshire-Farmer, 2023-10-25:
### Cura Version

5.5.0 - Beta 1

### Operating System

windows 11

### Printer

Ultimaker S5

### Reproduction steps

Load STL file (file loaded and schematic appeared on build plate as usual)
Set nozzle material to Tough PLA, with no support
Set slicing operation
Slicing failed with error message suggesting that I submit a error report

### Actual results

Nothing the design image remained on the build plate and slicing option re-appeared, when pressed the same results occurred

### Expected results

The sliced design should have become available to be transmitted to the printer.

I tried a variety of options, the only one that worked was to have the ultrafast print (0.3mm) with no support and no adhesion

### Add your .zip and screenshots here ⬇️

I tired to attach the .3mf project file but it was rejected as an unsupported  file type

GregValiant, 2023-10-25:
Thanks for the report.
Github rules are that anything other than a text file must be zipped.  We'll need that project file ("File | Save Project") to investigate.  If the file is deemed too large by Github you can post it on Google Drive (or something like it) and leave a link here.

Yorkshire-Farmer, 2023-10-26:
Dear Greg,

herewith the zip version of the project file

-- 
Dr. Peter Bailes
P & C. M. Bailes Farming
High Inhams Farm
Stillington Road
Sutton on the Forest
York
North Yorkshire
YO61 1EQ
Tel 01347-812589
Fax: 01347-811827
Mobile: 07796622921
Email: ***@***.***
GPS 54 °04' 39.0" N 1°06' 11.3" W
(54.0775°N,1.1030556 °W)
We are what we repeatedly do. Excellence, then, is not an act, but a habit.

GregValiant, 2023-10-26:
You have to post the zip file here directly.  It doesn't come through when trying to use Email.  It's a GitHub thing.

Yorkshire-Farmer, 2023-10-28:
[UMS5_Gully_Linear_Pattern_V3-Fillet001.zip](https://github.com/Ultimaker/Cura/files/13195062/UMS5_Gully_Linear_Pattern_V3-Fillet001.zip)

Yorkshire-Farmer, 2023-10-28:
Hopefully this has got through

Yorkshire-Farmer, 2023-10-28:
Dear Greg,

I have sent the file using the git hub link, hopefully it has arrived with
you in a usable form.

Best Wishes

Peter

-- 
Dr. Peter Bailes
P & C. M. Bailes Farming
High Inhams Farm
Stillington Road
Sutton on the Forest
York
North Yorkshire
YO61 1EQ
Tel 01347-812589
Fax: 01347-811827
Mobile: 07796622921
Email: ***@***.***
GPS 54 °04' 39.0" N 1°06' 11.3" W
(54.0775°N,1.1030556 °W)
We are what we repeatedly do. Excellence, then, is not an act, but a habit.

GregValiant, 2023-10-28:
I got it.
When Cura shows "red surfaces facing upwards" it indicates surfaces that are "inside out".  WHen the slicer encounters them it can get confused. 
Here is a view of the single rib.  This does not look right and is not repairable.
[image]

This is one of the short ribs and is also not repairable.
[image]

Here is that same short rib after repair by MS 3D Builder.
[image]

After moving the model on the Cura build plate it sliced, but I don't think this is what you had in mind.
This is the short rib
[image]

The model has non-repairable errors and so we can't tell if a bug is affecting slicing, or if the model is affecting slicing.

I see you have closed this report.  Is that what you wanted to do?  It's OK since there isn't anything we can do with this because of the bad model.

Yorkshire-Farmer, 2023-10-30:
Dear Greg,

Thank you for this analysis.  Having now printed this, I have also observed
the same thing, but it does leave a few points hanging.

The first is that the latest Cura software does manage to slice this
identical model, when the resolution is decreased to a 0.3mm feed (I have
not printed this), the second is that if I revert to the previous Cura
release (5.3), it does successfully slice the model at 0.1mm feed.

I tried this and also printed it and the result is fine except for these
support brackets, which , as you illustrate, do not come out as solid
supports.  They are rendered in Freecad as solid supports, but clearly
something is missing in the translation.  They were formed by creating a
"loft" between two identical closed 2D shapes attached to the
horizontal and vertical faces of the gulley cover.  I will do some more
research on what could have gone wrong in the configuration of the "loft",
but it is worth noting that Cura version 5.3 did allow the slice to
complete and also did allow the model to print to completion (albeit with
faulty support brackets).

I will let you know if I find a root cause at the freecad side of the
project or the "export" of the stl file, for your awareness in dealing with
other users.

In the mean time many thanks for your support and the effort that you have
devoted to understanding this problem.

Best Wishes

Peter

I

-- 
Dr. Peter Bailes
P & C. M. Bailes Farming
High Inhams Farm
Stillington Road
Sutton on the Forest
York
North Yorkshire
YO61 1EQ
Tel 01347-812589
Fax: 01347-811827
Mobile: 07796622921
Email: ***@***.***
GPS 54 °04' 39.0" N 1°06' 11.3" W
(54.0775°N,1.1030556 °W)
We are what we repeatedly do. Excellence, then, is not an act, but a habit.

GregValiant, 2023-10-30:
Lofts are always tricky for both the CAD software and the slicer to deal with.  The math of a lofted surface can be quite intense.
Since they are just gussets you may want to make them out of rectangles and then subtract a cylinder to create the radius.

The gussets don't appear to have been inset into the model.  As a consequence when they were merged there is a gap between the vertical leg of the gussets and the wall of the model.  It is always best to have overlap with (in this case) the gusset set into the model.  The overlap would disappear during the merge operation.  Trying to eyeball two surfaces to mate is tough unless the CAD package allows you to constrain the parts in all three axes.

In regards to the slicing - models like this that have a lot of geometric shapes on the build plate seem to be giving these last couple of versions of Cura some heartburn.  This is the third model in the last couple of weeks to come up with slicing problems.
I'm going to re-open this so the Cura team will have another example of this behavior to use in their troubleshooting.

Yorkshire-Farmer, 2023-10-31:
Dear Greg,

You are 100% correct.  I have never tried printing a "Loft" before, but I
found a tutorial on how to connect two different faces in a part design
space and thought it worth a try - big mistake.

I agree the old fashioned approach to extrude and pocket is much simpler
and safer.   Version 5 of the gulley cover will be designed that way..

With regard to the closure of the bug report, if i did it , it was
unintentional, I thought i was just uploading the project file to the
open report, so no problem in re-opening it  or using it for internal
education etc.

Hopefully it will help others.

Best Wishes

Peter

-- 
Dr. Peter Bailes
P & C. M. Bailes Farming
High Inhams Farm
Stillington Road
Sutton on the Forest
York
North Yorkshire
YO61 1EQ
Tel 01347-812589
Fax: 01347-811827
Mobile: 07796622921
Email: ***@***.***
GPS 54 °04' 39.0" N 1°06' 11.3" W
(54.0775°N,1.1030556 °W)
We are what we repeatedly do. Excellence, then, is not an act, but a habit.