Ultimaker/Cura #17106: Articulated Dragon 5.4.0

GigisZombie, 2023-10-25:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Ender 2 Pro

### Name abnormal settings

only changed quality .2/walls 4/inflill 10/speed 30

### Describe model location

i've tried rotating and moving will not slice

### Describe your model

i can slice with older versions of cura just not with 5.4.0

### Add your .zip here ⬇️

[Articulated Dragon Eryone PLA lw 0.4mm lh 0.2mm if 10% ext1 190C bed 60C.zip](https://github.com/Ultimaker/Cura/files/13169091/Articulated.Dragon.Eryone.PLA.lw.0.4mm.lh.0.2mm.if.10.ext1.190C.bed.60C.zip)

GregValiant, 2023-10-25:
Thanks for the report and the file.
Every dragon that comes through here is full of errors.  
5.4.0 does have a bug that can keep models from slicing.  You can try moving the model in Cura, rotate the model, or change the initial layer height.  Sometimes a combination is needed to fool the bug.  Because of all the errors you may not be able to trust the gcode if it does slice.
I'll mark this as a duplicate bug for now and go ahead and close this since 5.5 has been released as an update.  I hope you understand.

This sliced after I rotated the model 90°CCW.
[image]

closed 2023-10-25 by GregValiant

riaanp2012-ship-it, 2026-11-05:
this one is the best accoring to me https://makerworld.com/en/models/3551279-articulated-dragon

GregValiant, 2026-11-05:
A few years ago someone posted a model of a "Crystal Dragon" that was articulated.  It had a ton of errors in it.  That makes for a tough slice.  Somehow the designer was notified of the problems and they were able to correct it.  I haven't seen it show up here since the model was fixed.
This is the error report on the Crystal Dragon.
-> Analyzing file
--> 115 Naked edges  (?)
--> 8 Planar holes  (?)
--> 15213 Non-Manifold edges  (?)
--> 23751 Inverted normals  (?)
--> 100 Duplicate faces  (?)
--> 407 Degenerate faces  (?)
That's a grand total of 39,594 errors.  Not good.
Today, it sliced in both Cura and PrusaSlicer.  Three years ago it wouldn't slice in any of them.