Ultimaker/Cura #17110: Support 3MF Core spec for object.name on import and renaming objects

ansonl, 2023-10-26:
### Is your feature request related to a problem?

Cura adds [object.name](https://github.com/3MFConsortium/spec_core/blob/master/3MF%20Core%20Specification.md#chapter-4-object-resources) for each object when exporting in 3MF. The name that Cura sets is the original imported filename incremented with a number for each additional object in the file like Filename, Filename (1), Filename (2), etc. However, Cura ignores object.name attribute in imported 3MF files and overwrites them with the generic filename on export.

Since the object.name attribute is used to name individual objects, it would be useful if Cura named imported objects according to the existing object.name if already set as well as allowed renaming objects so that exported 3MF files would have contextual object names that users set. 

### Describe the solution you'd like

Cura could have a right click option or editable textbox on objects in the object list for users to set object names.

Cura should use the existing or user specified object name for object.name in saved 3MF files.

### Describe alternatives you've considered

Object naming through object.name is already supported in Prusaslicer and Blender (through the blender 3MF addon)

As seen in the Prusaslicer screenshot below for a [test multiobject print](https://www.printables.com/model/228038-ninja-pot-01), individual objects in the model 3MF file have descriptive names that give the users context what an object is. 

[image]

### Affected users and/or printers

All users using 3MF files. All users with dual or multi extrusion. 

### Additional information & file uploads

Test 3MF file with two objects with different names [DC-linear-dual-v2.zip](https://github.com/Ultimaker/Cura/files/13172565/DC-linear-dual-v2.zip)

Another test 3MF file with multiple named objects and colors specified: https://www.printables.com/model/228038-ninja-pot-01

MariMakes, 2023-10-26:
Hey @ansonl,

Thanks for your report 👍
I've never worked with 3mf files in that way, so I wasn't aware that this is how people use them. 
But your request makes sense if you do. 

I'll bring it up to the team to see what they can do to improve it.
But I expect that if it's not a quick fix it will not get a high priority to implement. 
Fingers crossed 🤞

pietchaki, 2023-11-24:
I believe that this improvement I made solves the same issue from @ansonl :
https://github.com/Ultimaker/Uranium/pull/898#event-10832695750

ansonl, 2023-11-29:
@pietchaki Do you know if your improvement is in the latest Cura 5.5 so I can try? I'm still using 5.4 due to a [custom printer profile](https://github.com/Ultimaker/Cura/pull/16001) that has not been merged.