Ultimaker/Cura #17111: Cat

terrystanley1, 2023-10-26:
### Cura Version

5.5.0

### Operating System

windows 11

### Printer

Ender 3 S1 pro

### Name abnormal settings

No, you need to change any setting before it slices

### Describe model location

No

### Describe your model

Yes, they all worked with 5.4.0. This problem now affects all my models since installing 5.5.0

### Add your .zip here ⬇️

[3-Cat.zip](https://github.com/Ultimaker/Cura/files/13173632/3-Cat.zip)

GregValiant, 2023-10-26:
Thanks for the report.
You didn't actually mention what the problem is.  Your STL sliced fine for me.
[image]

terrystanley1, 2023-10-26:
Hi,I changed from 5.4.0 to 5.5.0 yesterday. Since then, whenever I open a model, it will not slice unless I change a setting such as profile, fill etc. I get the error message that it was unable to slice, but no errors are specified. If I then change a setting and then change the setting back again, it works fine. This happens on all my models, which I have loaded on earlier versions of Cura without problem.Thanks

GregValiant, 2023-10-26:
In Cura - repeat the process that causes the behavior and then use the "Help | Show Configuration Folder" command and locate the "Cura.log" file.  Zip the log file and post the zip file here.  The Cura team will need it to chase down what is going on.

terrystanley1, 2023-10-26:
Hi,

 

There doesn’t appear to be a .log file, but I am wondering if either of these are what you are looking for. I also include a screen shot so that you can see the error message. If there is anything else I can supply, please let me know.

 

Meanwhile, I have reverted to 5.4.0

 

Thanks