Ultimaker/Cura #17112: Treesupport Brim

InjectedPie, 2023-10-26:
### Is your feature request related to a problem?

When dealing with larger prints, I encounter an issue where individual tree support branches sometimes detach, even though the print itself is large enough not to require adhesion (brim). What if there were an option for an 'inside brim' for the tree support, akin to an hollow elephant's foot, to improve its adhesion to the bed?"

### Describe the solution you'd like

 'inside brim' for the tree support, 

### Describe alternatives you've considered

Using a regular brim works, but for larger prints, it results in a significant filament wastage.

### Affected users and/or printers

Anycubic vyper, with Klipper and stealthburner Mod

chipmunk1977, 2023-10-26:
As I understand it, that is exactly what "Enable Support Brim" does!  It produces a brim inside the tree supports.

GregValiant, 2023-10-26:
I agree.  The "Support Brim" and "Support Brim Width" would seem to cover this.  There was a bug in 5.4.0 that would keep the support brim from developing if the bed adhesion was set to "None".  That has been fixed in 5.5.0
Support Brim is always on the inside of the support trunk.  This is with a Skirt.
[image]

This is 5.5.0 with no bed adhesion.
[image]

If your nozzle starts hitting the support structure as it gets taller, then that's a mechanical issue but it can certainly knock over a support.  Enabling Z-hops is the usual fix for that and Avoid Supports When Traveling can also help.

InjectedPie, 2023-10-26:
So thats a bug thanks for letting me know... 
Just switched from Superslicer to Cura 5.4 ill update to 5.5 I was wondering about the Support trim doing nothing on 1 print and something that looked like  (and probably was) a layershift on another print

InjectedPie, 2023-10-26:
[image]
works as it should in 5.5 thx anyway

closed 2023-10-26 by InjectedPie