Ultimaker/Cura #17116: Deceleration at corner radius

helpMe4332, 2023-10-26:
### Is your feature request related to a problem?

Cura doesn't recognize corners with a radius value above 0, and the print head makes a 90° turn at full speed. 
For example, a corner with a radius of 2 mm. Jerk and acceleration settings won't apply to this corner, resulting in print quality issues. 

### Describe the solution you'd like

Useful settings:
- Minimum radius (to what radius the settings are applied: for example, if the set value is 1 mm, speed settings will apply to any corner radius of 1 mm or less)
- Minimum corner angle (to what angle settings are applied: for example, if the set value is 85 - 95°, speed settings will apply to any corner within the set angle value)
- Deceleration before corner (reduce speed as nozzle approaches corner)
- Deceleration distance (distance at which deceleration applies)
- Corner speed (speed the nozzle travels during corner move) 
- Acceleration (acceleration after corner)
- Acceleration distance (distance at which acceleration applies after corner)

This will solve print quality issues, especially at higher speeds, and reduce wear on mechanical parts. 

### Describe alternatives you've considered

The only alternative is slowing down overall speed. 
Which isn't a solution - some filaments require minimum speeds.

### Affected users and/or printers

All printers and users.

GregValiant, 2023-10-26:
_"...a corner with a radius of 2 mm. Jerk and acceleration settings won't apply to this corner..."_  I've never heard that.

What printer are we talking about here?  My Ender always uses Accel and Jerk control for everything.  If I enable Accel and Jerk in Cura then I can select settings for different features.  If I don't enable Accel and/or Jerk in Cura then the printer defaults to what is in it's M201, M204, and M205 registers.  They are also editable by sending gcode commands.
This is from a response to M503 by my Ender 3 Pro:
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000 Y3000 Z100 E1000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P1000.00 R1000.00 T1000.00
B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.40 E5.00
So in regards to my printer there are no situations where Accel and Jerk is "Disabled".  Accel in particular cannot be "0" and will never be "Infinite".  I am not able to turn Accel and Jerk control off at the printer and I believe that would be true for most printers as that would result in "Infinite" acceleration (or at least as high as the particular steppers could manage).

helpMe4332, 2023-10-26:
I'm using Guider II by Flashforge, Marlin.
But the problem seems to be with Cura. 
For one, it doesn't show the speed change on "color scheme: speed". Although it might not be a feature. 
The problem is more apparent with prints. 
A print with no corner radius prints perfectly. 
A print with a small corner radius has overshooting throughout the circumference of the part, especially at corners. 

Adjusting Jerk/Accel has no effect on the quality, even when reduced to 5 mm/s jerk, 500 mm/s² accel. 

Here's what Cura 5.5 is outputting: 

;TYPE:WALL-OUTER
M900 K0.750000 T0 ;added by LinearAdvanceSettingPlugin
;MESH:Body1.stl
G1 F3300 X-6.875 Y-10.044 E648.32593
...
G1 X-7.105 Y-9.983 E652.84659
M204 S1500
M205 X12 Y12
G0 F7200 X-6.875 Y-10.044
...
G0 X-11.415 Y-13.175
M204 S1000
M205 X9 Y9
G1 F3300 X-10.874 Y-13.439 E652.91529
...
G1 X-11.542 Y-13.046 E658.49545
G0 F2970 X-11.415 Y-13.175
M204 S1500
M205 X12 Y12
G0 F7200 X-10.874 Y-13.439
...
G0 X-14.355 Y-14.494
M204 S1000
M205 X9 Y9
G1 F3300 X-14.675 Y-14.322 E658.53677
...
G1 X-14.14 Y-14.546 E667.54847
G0 F2970 X-14.355 Y-14.494
M204 S1500
M205 X12 Y12
G0 F7200 X-14.675 Y-14.322
...
G0 X-12.794 Y-13.394

GregValiant, 2023-10-27:
That looks fine.
The printer does nothing without being told.  Once a setting is made it stays in place until something changes it.
M204 S1000
M205 X9 Y9
sets the Accel to 1000 and the Jerk to 9.  It remains in effect and is used on _**every single move**_ the printer makes until the next M204 and M205 lines change the values.  It doesn't need to be constantly repeated and it is never ignored.
If you turn off Accel and Jerk control in Cura the printer will use what it has in M204 internally _**for every single move**_ in the gcode file.  All of them.
Remember too that there are no curves arcs or circles in Gcode.  There are just short line segments that approximate a curve, arc or a circle.

helpMe4332, 2023-10-27:
My guess is either: 
1. Cura isn't recognizing small radii as corners. Ex: a radii of 50 mm can easily handle a turn at 60 mm/s. But a radii of 2 mm is too sharp. Both have a radii, but require different speeds. 

2. The radius path (2 mm) is too short - the printer speed change is delayed too close to the corner. And never reaches 10 mm/s (jerk). Ex: travelling at 60 mm/s approaching the corner, sharply decelerates to 10 mm/s, which ends up being a split second (if at all). It may never reach the jerk value. 

I'll have to test an accel of 100 and jerk of 1. 
Otherwise, there's still no explanation for why a corner radius results in defects, and a sharp corner is perfect.

GregValiant, 2023-10-27:
This model is my own calibration shape.  It has a variety of radii including an inside 1mm and 2mm and an outside 1mm and 2mm.  This screenshot is of the 2mm inside radius.  Each line segment that comprises the wall is 0.31mm long.  My printer zips right through all of this without any hesitation.
[image]

Does you firmware support Linear Advance or some other pressure adjusting feature?   Is it possible that the nozzle pressure is not adjusting fast enough on a sharp corner?
I forgot to ask what kind of Print Speeds we are talking about?

I printed the model at 75mm/sec for everything, Accel at 1000 for everything and Jerk at 10 for everything.  With 7% infill and two walls its a 34minute print.
Yellow marks are bumps and sharp corners.  Blue marks indicate the designed radius.
[GV_CalibrationShape.zip](https://github.com/Ultimaker/Cura/files/13184370/GV_CalibrationShape.zip)

Nozzle movement was CW and there is a 0.20-0.25mm bump just as the nozzle goes into the turn.  There is a second bump at the end of the short run.  It appears to me that there is oozing while the nozzle is at the sharp corners.  The sharp point on the top is indeed designed as a point but it is 1.25mm wide while my line width was 0.45 so there appears to be some oozing there as the printer does come to a complete stop to reverse course.
My 1.1.8 Marlin firmware doesn't support Linear Advance or other pressure regulating settings.

[image]

helpMe4332, 2023-10-27:
I'm using Cura and linear advance is enabled, but can't confirm if the printer's responding to it. 
The material overshooting is throughout the part, which leads me to believe it's not an extrusion issue. 

All of my tests were printed at 60 mm/s. 
Based on what I can see with your part, I think you're having a similar problem, where the speed isn't being reduced at small rounded corners. It's less evident with inner corners, as the nozzle goes into the part, as opposed to away from the part. 

Id be curious to know, between a 1 mm, 2 mm, 3 mm, 4 mm, etc. corner radius. When does Cura slow down (apply jerk), and does it factor in the wall speed. 
It's like driving a car and taking a sharp turn at 90°, vs a slowly rounded bend. The cars speed should decelerate proportionally to the radius of the road. 

Both parts printed with the same settings. The only difference is the rounded corners:
[image]

GregValiant, 2023-10-27:
Once Cura figures out the line width it wants for a particular extrusion it's simply LineWidth x Layer Height x Length of extrusion to get the E number.  Speed is whatever it was supposed to be for the feature, and then the printer applies the Accel and Jerk whether it's from it's defaults, or from the gcode file.
The only way to make an adjustment would be to split the line into segments (which is what "Coasting" does) or to utilize Linear Advance (provided the printer can use it).
The error in the corners of my print are about 0.25mm.  Given my line width of .45 that's about a 50% overshoot.  

If it was possible for the printer to apply an Accel to the Extruder at the start of an extrusion, and a different Decel to the Extruder at the end of the extrusion then that might have some possibilities.  That would need to be applied in the firmware.

I don't know if it is even possible to build a "look ahead" into Cura to pre-plan the end of an extrusion.  It would seem to be a ton of work.

helpMe4332, 2023-10-27:
Cura could calculate the relative flow rate and speed, based on a few simple inputs. 
Wall angle: as the angle increases or decreases from 90°, the flow and speed decrease proportionally 
Radius: as the radius increases from 0, the flow and speed increase proportionally

Coasting and linear advance would see the same problems, if Cura can't recognize a corner. If a 0 mm radius = corner, but 1 mm radius = regular path, it won't help. 
In most of my tests, coasting and linear advance were on.