Ultimaker/Cura #17119: Slice fails with support turned on, but works without support

ericjforman, 2023-10-26:
### Cura Version

5.5.0

### Operating System

Mac OS 12.7

### Printer

Ultimaker S5

### Name abnormal settings

Slicing works with Default Draft, but fails when Support is turned on. 

### Describe your model

Yes, watertight. Mesh Fix reports no issues. Rebuilding normals had no effect.

### Add your .zip here ⬇️

[UMS5_DE_Houdini test03 WB-thickened 2mm shell with holes.3mf.zip](https://github.com/Ultimaker/Cura/files/13182502/UMS5_DE_Houdini.test03.WB-thickened.2mm.shell.with.holes.3mf.zip)

Sometimes slicing runs all the way through but does not report an error, it just shows nothing (after a 15+ minute slicing process), and all I can do is Slice again.

HellAholic, 2023-10-27:
Hey @ericjforman ,
I have a quick workaround for you, if you slightly rotate the model and move it to so that is not centered in the build plate the slice finishes. Photo below for reference. 
[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic