Ultimaker/Cura #17128: Tenlog TL-D3 Pro does no longer recognize files sliced files after updating to Cura 5.5.0

lrisRain, 2023-10-27:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

Tenlog TL-D3 Pro

### Reproduction steps

1. I sliced the file like usual and ejected the sd card.
2. I inserted the sd card into the sd card slot of the Tenlog TL-D3 Pro
3. I tried to select the file to print it.

### Actual results

There are no results shown on the SD card despite 2 sliced files being aviable. Files sliced with 5.4.0 still get recognized.

### Expected results

Sliced files should be able to be selected for printing on the Tenlog TL-D3 Pro.

### Add your .zip and screenshots here ⬇️

[Castle-2F_right3.zip](https://github.com/Ultimaker/Cura/files/13188600/Castle-2F_right3.zip)

lrisRain, 2023-10-27:
Can be closed. It appears it is an issue with the SD card itself. The second sd-card (same model) with the very same files works with the printer.

GregValiant, 2023-10-27:
Nice catch.
I'll go ahead and close this.

closed 2023-10-27 by GregValiant