Ultimaker/Cura #17129: Small Top/Bottom Needs to go from outer to inner - 5.5 bug

PhilBaz, 2023-10-27:
### Cura Version

5.5

### Operating System

win 10

### Printer

Voron 2.4

### Reproduction steps

Slice using Small Top/Bottom 

### Actual results

 Small Top/Bottom  starts path over air, rather than anchoring to wall.

[image]

### Expected results

Anchor to wall rather than print over air.

### Add your .zip and screenshots here ⬇️

[dino1.zip](https://github.com/Ultimaker/Cura/files/13190737/dino1.zip)
[image]

GregValiant, 2023-10-27:
Thanks for the report but without a project file there is no way to investigate this.

PhilBaz, 2023-10-27:
@GregValiant 

It's in the zip. Above the image.

PhilBaz, 2023-10-27:
@GregValiant See it?
[image]

GregValiant, 2023-10-27:
My bad...thanks.
I can reproduce this with the "Small Top Bottom Width" at 5.0 but the box is yellow indicating a warning.  The warning is displayed when the setting is above 1.5.  With that setting at 1.5 (and below) the problem goes away.  Is there a reason you have 5.0 in there?
This looks like a setting issue rather than a bug as you are out of the recommended zone.
This is at 1.5.
[image]

PhilBaz, 2023-10-27:
Well, at 1.5 the function hardly catches any narrow areas that would produce the violent back and forth the feature is described as being intended to reduce.

Look at the dino tail point. At 5 it catches a decent area and turns it to concentric. That's good, no/less micro lines shaking the printer back and forth.

[image]

However the concentric area, starts at the center, rather than at the wall where it would be anchored. This makes its otherwise useful implementation fail....

[image]

I'd like to use this at larger values, it would offer more benefit at larger values! I think this feature originally comes from Orca slicer, and the implementation there works as I describe. If you are limiting this feature to areas below 1.5mm I would arege you are merely limiting the range over which it is beneficial. In orca it operates over a much larger range for non-top surfaces. maybe 5mm-10mm.

Key is having the concentric section start at the wall so its anchored.

GregValiant, 2023-10-27:
If it is going to do concentric then it likely needs to plan on some support inside.  Maybe they should make the infill denser in those areas.  That way it wouldn't start over air.  Right now all it will do is puke that concentric area down into a gap in the infill.  It might help if it went outside to inside but I see that the "Extra Skin Wall COunt" is set to 0 but it's putting one around that little area and it goes in last.  If it was there first and movement was outside to inside then it might help keep the plastic in place.