Ultimaker/Cura #17133: Infill Linfe Multiplier prevents slicing big parts

BackwardEcho, 2023-10-28:
### Cura Version

5.5

### Operating System

Windows 10

### Printer

modded Tronxy X5SA-400

### Reproduction steps

Place large part on build volume
Infill : ~30%+, it works at 25% but takes a while to slice.
Infill Line Multiplier : 2

I don't think infill pattern matters as it also failed with gyroid

### Actual results

Slicing gets stuck around 50% with 100% CPU utilization

### Expected results

not that

### Add your .zip and screenshots here ⬇️

[Issue.zip](https://github.com/Ultimaker/Cura/files/13194491/Issue.zip)

Switching 'Infill Line Multiplier' to 1 makes it all work just fine.

GregValiant, 2023-10-28:
Thanks for the report and the file.
When you hover the mouse over the label "Infill Line Multiplier" the tooltip explains that the setting is affected by "Connect Infill Polygons".  With "Connect Infill Polygons" turned off, the project sliced.
[image]

So I don't see a bug here as the situation seems to be addressed with that tooltip and changing the setting allows it to slice.

I use the "Infill Line Multiplier" on a lot of models.  Grid Infill with the Infill Line Directions at [90] and the multiplier at 3 creates a very robust rib system within a print.
This is my preference for _heavy duty_ infill with Grid at 10% and the Infill Line Multiplier at 3.  With your other settings (infill at 20mm/sec and Walls at 15mm/sec) this is a 48hr print.
[image]

So the behavior you are seeing appears to be explained in the ToolTip and I'm inclined to remove the bug label and close this as a "setting issue" rather than a bug.  Let us know what you think.

(BTW  Your setting profile says "ABS" but your material is "PLA".)

BackwardEcho, 2023-10-28:
Thank you for your reply.

I guess I incorrectly titled the issue then, It wasn't necessarily a
problem with not getting the model to slice at all not matter what, but
that it appears to get stuck and the CPU usage ramps up the entire time it
tries with a line multiplier of 2 and "Connect Infill Polygon". I've let it
go for about 10 minutes or a bit more before with no noticeable change in
completion but CPU was still at 100%. If I left it for a few hours would it
have completed the slice?

I am aware of the different profile and material, I tend to ignore the
material and change the profile as I need to match the actual material and
have only recently started messing with the material settings in Cura. So
far, only my hotend temp changes with it while my bed temp stays according
to the profile with no "f(x)" icon. (While I am thinking about it, and I
know this isn't the place, what of putting the "Scaling Factor Shrinkage
Compensation" setting in with the material profile print settings?)

GregValiant, 2023-10-28:
In regards to slicing time, there are certain models that have a lot of geometric shapes on the build plate that take forever to slice.  That has come up several times here lately in bug reports.  The workaround for that seems to be to disable "Optimize Wall Printing Order".  For those situations it can dramatically affect how long it takes to complete a slice.  Whatever the problem is there - you might have found something similar with the "Connect Infill Polygons" setting and the Multiple Infill Walls setting.

In regards to the "Scale" settings they are in with the other material settings but they are not linked to a material.  My guess would be that in general the amount of shrink is dependent on the material but it's also heavily influenced by the way you set up a slice.  Things like the type of infill and the density can have a profound effect.  Some types of infill (like gyroid) would have almost no effect as it has no inherent strength.  My Heavy Grid infill would have a much bigger effect on shrinkage.  Then there is the Z shrinkage as compared to the XY shrinkage.  I have to guess again and my guess would be that very few materials suffer from shrinkage in the Z due to the nature of FDM where parts made in a mold would suffer equally on all three axes.