Ultimaker/Cura #17137: UMS5_Flaming Cauldron

skybernd, 2023-10-28:
### Cura Version

5.5

### Operating System

MAC OS 

### Printer

Ultimaker S5

### Name abnormal settings

none 

### Describe model location

no 

### Describe your model

Not With Cura 

### Add your .zip here ⬇️

[UMS5_Flaming Cauldron.3mf.zip](https://github.com/Ultimaker/Cura/files/13195887/UMS5_Flaming.Cauldron.3mf.zip)

GregValiant, 2023-10-28:
Thanks for the report and the file.
The model doesn't appear to have any errors and I can't get it to slice either.  It did slice (took a long time) with my Ender as the active printer so this might have something to do with settings.

The Cura team will take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic