Ultimaker/Cura #17138: slicing freezes for specific model

pipyjon, 2023-10-28:
### Cura Version

5.5.0

### Operating System

windows 11

### Printer

anycubic mega x

### Reproduction steps

[gnome shell.zip](https://github.com/Ultimaker/Cura/files/13195900/gnome.shell.zip)
stood this model up with the thicker part to the bed. waited over night for it to slice with tree supports at 5% cubic infill at 60 speed 85 fan speed.

### Actual results

slicing froze about 1/3rd of the way. no error message.

### Expected results

slicing should have finished

### Add your .zip and screenshots here ⬇️

[AI3MX_gnome shell.zip](https://github.com/Ultimaker/Cura/files/13195926/AI3MX_gnome.shell.zip)
[image]

GregValiant, 2023-10-28:
Thanks for the report.
You didn't include a project file so I don't know your settings.  This took about 11 seconds to slice using an Anycubic Mega X printer and the "Normal" profile.  So I'm unable to duplicate the problem you are having.
I do get the "model is not watertight" warning so I don't know if I would trust the slice.

pipyjon, 2023-10-29:
are you saying the link above the screenshot isnt the project file? i ended up having to delete my printer profiles and create new profiles. it worked after that.

closed 2023-10-29 by pipyjon

GregValiant, 2023-10-29:
The first one (gnome shell.zip) was just the STL and once I saw a file, I didn't look for another.

This is the STL file.  The red surfaces are errors in the model.
[image]

This is the project file.  The errors look like they have moved but it's still a lot of errors.
[image]

Even after repair the model doesn't look right.  The model from the project file is on the left and the "repaired" model is on the right.  Something is still wonky with the model.
[image]