Ultimaker/Cura #17140: Problem with {initial_extruder_nr} at Multiple extruder

RS-3D, 2023-10-28:
### Cura Version

5.5.0

### Operating System

Windows 10

### Printer

Custom Printer with 4 Toolheads

### Reproduction steps

I´m Slice the Modell for example with Toolhead 3 (T2).

### Actual results

In the final G-code file it starts correctly with the start up, i.e. with setting T2. However, when initializing the temperatures for the respective print head, extruder0 is always selected instead of extruder2. This has been the case since version 5.4, everything was OK in 5.3.
A cutout from G-Code result: (extruder2 should be at extruder0)

;Generated with Cura_SteamEngine 5.5.0
T2
M82 ;absolute extrusion mode
ACTIVATE_EXTRUDER EXTRUDER=extruder0
M140 S80.0 ; set bed temp ; set bed temp
;M190 S80.0 ; wait for bed temp
M109 T0 S220.0 ; wait for extruder temp
G90 ; use absolute coordinates
G92 E0
G1 E30 F200
START_PRINT
START_GO
G92 E0.0
PICK_extruder0

### Expected results

Please change the {initial_extruder_nr} to work correctly again in the next version.

Otherwise I can no longer use cura with peace of mind.

### Add your .zip and screenshots here ⬇️

G-Code Cutout from Start:
;FLAVOR:Marlin
;TIME:5333
;Filament used: 0m, 4.72721m, 0m, 0m
;Layer height: 0.2
;MINX:126.447
;MINY:171.883
;MINZ:0.3
;MAXX:233.55
;MAXY:298.117
;MAXZ:8.3
;TARGET_MACHINE.NAME:Unknown
;Generated with Cura_SteamEngine 5.5.0
T1
M82 ;absolute extrusion mode
ACTIVATE_EXTRUDER EXTRUDER=extruder0
M140 S80.0 ; set bed temp ; set bed temp
;M190 S80.0 ; wait for bed temp
M109 T0 S220.0 ; wait for extruder temp
G90 ; use absolute coordinates
G92 E0
G1 E30 F200
START_PRINT
START_GO
G92 E0.0
PICK_extruder0
G92 E0
G92 E0
G1 F1200 E-3
;LAYER_COUNT:41
;LAYER:0
M107
G0 F3600 X127.459 Y183.654 Z0.3
;TYPE:SKIRT
G1 F1200 E0
G1 F1800 X128.191 Y182.867 E0.05362

GregValiant, 2023-10-29:
Thanks for the report.
I have seen this myself.  Take a look at #15673 and #16715 and #11985 .
I see that on Sept 18 @MariMakes added a ticket to the backlog [CURA-11053](https://ultimaker.atlassian.net/browse/CURA-11053)
I'll mark this as a duplicate.  I'll leave it open and the Cura team will take another look.

RS-3D, 2023-10-29:
Thanks Greg.
I hope the cura team can fix the problem quickly. I have to manually check and edit the G-code after each slice.

GregValiant, 2023-12-11:
A little bit of an update:
The problem seems to be that if an extruder isn't:
- Being used in the print
- Is disabled (particularly in the case of E1/T0)

It may still be listed as an "Adhesion Extruder" or a "Support Extruder" as those setting fields don't get updated for either of those cases.  So the "initial_extruder_nr" may report the wrong number even when it seems obvious that an extruder is unused or disabled because "Build Plate Adhesion" and/or "Supports" print first on a layer.  When disabling an extruder, both of those settings are hidden so "out of sight out of mind".  That would be fine if the settings they contain were correct.

A workaround is to manually change the "Adhesion Extruder" and the "Support Extruder" to viable numbers.  That will allow "initial_extruder_nr" to return the proper number.