Ultimaker/Cura #17141: Slicing Failed

jcford73, 2023-10-28:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

Sovol SV06 or Elegoo Neptune 3 Pro

### Name abnormal settings

None. Does not slice with default profile.

### Describe model location

No

### Describe your model

Grim Reaper figure from Mimetics3D (via Thangs.com)

### Add your .zip here ⬇️

[SS_ReaperCloak_Main.zip](https://github.com/Ultimaker/Cura/files/13196906/SS_ReaperCloak_Main.zip)

GregValiant, 2023-10-29:
Thanks for the report.
I can't get this to slice either.
Someone from the Cura team will take a look.  This is another organic shape that @jellespijker may be interested in.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic