Ultimaker/Cura #17145: cura 5.5.0 does not connect to printer  with usb old version 5.3.0 has no problems connecting

ggrrf667, 2023-10-29:
### Cura Version

5.5.0

### Operating System

windows 11

### Printer

anycubic neo

### Reproduction steps

1 turn on printer 
2open new cura
3 only print via card 
4no option to print via usb
5 open cura 5.3.0
6 instantly detects printer

### Actual results

not able to detect printer via usb cura 5.5.0

### Expected results

cura should have detected printer as older cura does

### Add your .zip and screenshots here ⬇️

no files to submit

GregValiant, 2023-10-29:
Thanks for the report.
There are currently two plugins for USB Printing available in the Cura Marketplace.  There were numerous complaints regarding the "auto-enable" feature and conflicts with other USB devices (besides printers) that people had plugged into their computers.  It's possible that it was decided not to have USB Printing enabled by default.  I always disable it as a matter of course and I can't remember if I had to disable it with 5.5.0.

Go to the Cura Marketplace and select the "Gear" icon.
Scroll down to USB Printing and see if it is enabled.  That's the version that was enabled by default in previous Cura versions.
A revised plugin is "USB Printing NG" that is available under "Plugins.  You might try that one.  Don't enable them both as they might conflict with each other.
I just checked with my Ender and both the older "USB Printing" and "USB Printing NG" connect as expected.

There will always be issues with printing over the USB.  There are so many hardware, software, and firmware combinations out there that it's impossible to account for all of them.  In addition since no UltiMaker printers use USB printing they aren't going to pay their people to fix it for 3rd party printers.  That leaves it up to members of the open source community.  Essentially, It either works for you or it doesn't.

fieldOfView, 2023-10-29:
>  It's possible that it was decided not to have USB Printing enabled by default. 

That is not the case. It would definitely be in the release notes if it were.

ggrrf667, 2023-10-30:
I tried again  and it is working now I did nothing on my end but glad it is working now thanks for replying before