Ultimaker/Cura #17146: The model is not cut with the standard settings made by me. I tried to put the manufacturer's profile there, the model was cut.

som1235, 2023-10-29:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Anycubic Kobra 2 Neo

### Describe your model

The model is not cut with the standard settings made by me. I tried to put the manufacturer's profile there, the model was cut.

### Add your .zip here ⬇️

[ReaperCloak_Main.zip](https://github.com/Ultimaker/Cura/files/13198142/ReaperCloak_Main.zip)

DanWSDev, 2023-10-29:
Hi @som1235 within your zip file it just an STL file and does not have your printer settings included. A way to send your printer settings is by making sure it is in an.3mf format. How to do this save the project and send that folder so it includes your printer settings. The model seems to load in Cura 5.4.0 fine on my settings but probably very different to yours can please include them or list them.
[image]

som1235, 2023-10-30:
> Привет, @som1235, в вашем zip-файле это просто файл STL, в который не включены настройки вашего принтера. Способ отправить настройки вашего принтера - убедиться, что они в формате.3mf. Как это сделать, сохраните проект и отправьте эту папку, чтобы в ней были указаны настройки вашего принтера. Модель, похоже, загружается в Cura 5.4.0 нормально в моих настройках, но, вероятно, сильно отличается от ваших, пожалуйста, включите их или перечислите. [image]

THANK YOU, IT HELPED