Ultimaker/Cura #17148: Profile saving is not possible anymore sinds version 5.5.0

roel8032, 2023-10-30:
### Cura Version

5.5.0

### Operating System

W11

### Printer

Anet A6

### Reproduction steps

In version 5.4.0 it was possible to save profieles.
Since 5.5 not anymore

### Actual results

No way to save profiles

### Expected results

No way to save profiles

### Add your .zip and screenshots here ⬇️

No

GregValiant, 2023-10-30:
Thanks for the report.
When you have made changes to a setting profile and select "Manage Profiles" at the bottom of the profile list, you should see this dialog come up.
[image]

If there have been no changes between the current settings and the profile then the "Update Profile" and "Discard Changes" buttons will be greyed out.  "Create New" should be available.

What comes up when you select "Manage Profiles"?

roel8032, 2023-10-30:
Met vriendelijke groeten,

 

Roel Kisteman

06-22.25.44.80

 

 <https://user-images.githubusercontent.com/64202104/279006364-9c2d1487-31d6-4232-bb59-121f9546157c.png>

GregValiant, 2023-10-30:
This is with "No Changes" to the selected profile.
[image]

This indicates that changes have been made to the selected profile and the buttons should be visible within the drop down box.
[image]