Ultimaker/Cura #17150: Unecessary supports added - supports not close to the print

Adammat, 2023-10-30:
### Cura Version

5.5.0

### Operating System

Windows 10 Education

### Printer

Ender 3

### Reproduction steps

1. I uploaded my print
2. I added support to the print
3. Set support to normal, and touching Buildplate
4. I sliced it

### Actual results

The slice made support that did not even close to connect to the print

### Expected results

The slicer should have made apporopiate supports that helps the print and no unecessary supports

### Add your .zip and screenshots here ⬇️

The support does not touch the print
[image]
These are my support settings
[image]
This is the file I'm trying to print
[Now_Playing_-_Vinyl_Holder_4890985.zip](https://github.com/Ultimaker/Cura/files/13204155/Now_Playing_-_Vinyl_Holder_4890985.zip)
Here is my configuration folder:
[5.5.zip](https://github.com/Ultimaker/Cura/files/13204189/5.5.zip)

GregValiant, 2023-10-30:
Thanks for the report.
I'm not sure how you have it oriented I I don't see those slots anywhere.
Use the "File | Save Project" command, zip the 3mf file, and post the zip folder here.
This doesn't look much like what you came up with.
[image]

Adammat, 2023-10-31:
Here you go
[CE3PRO_Now_Playing_-_Vinyl_Holder.zip](https://github.com/Ultimaker/Cura/files/13214561/CE3PRO_Now_Playing_-_Vinyl_Holder.zip)

GregValiant, 2023-10-31:
Can I close this?  I don't see a problem with either Tree or Normal supports.
[image]
[image]

Adammat, 2023-11-01:
Try changing it from Support placement "Everywhere" to "Touching Buildplate": If the problem still dosn't occur, you can just close it.

GregValiant, 2023-11-01:
In that first image - did you happen to have "Use Towers" enabled?  If so then they weren't meant to touch the model.  The project file you posted does not have "Use Towers" enabled.

In regards to Towers - with the "Support Horizontal Expansion" at 0.8 and the "Support X/Y Distance" at 0.8 the Tower grew for the support structure within the holes, but didn't connect to them because the distances canceled each other out.

If all that is correct, then the result was "expected behavior".