Ultimaker/Cura #17151: Support stability

mikeywaldron, 2023-10-30:
### Is your feature request related to a problem?

When running a long print. Everything is going smoothly then one tree fails which means the entire print needs to be stopped.

### Describe the solution you'd like

Ideally, set a brim inside and outside of support base. This would increase adhesion on the area making it less prone. Personally I've tweaked the trees with bigger trunk diameter etc... Brim the entire model adds this but requires more clean up brim only supports would be great!

### Describe alternatives you've considered

Another possibly would be to allow a number of base layers for the inner brim of trees, making this more dense would increase adhesion, so being able to set say 3-5 base layers on a brim inside the tree could improve adhesion

### Affected users and/or printers

I'm using a ender 2 V2 modified, running klipper. Tree supports have been tuned with trunk diameters and appropriate organic support tests have been tuned and passed

GregValiant, 2023-10-30:
I always thought it was odd that the size of the tree support brim was limited by the size of the trunk.
You can set a "brim distance" so the brim does not become part of the print but unfortunately it doesn't become part of the support either.

There are a couple of things that come into play with this.
- Bed slinger printers are tough on the bed adhesion.  When the bed is slamming back and forth the print wants to break loose.  As a print gets taller and the upper portion gets heavier the situation gets worse.  There is a new post processor (Limit the XY Accel and Jerk) to address that by limiting the Accel on particular axis.  For example you can gradually drop the Y accel over a range of layers while keeping the X accel at whatever it was.
- Enabling Z-hops will keep the nozzle from whacking any little protrusions that might be sticking up from the support structure.
- Enabling "Avoid Supports When Traveling" can also help.  Setting the Avoid Distance to 1 or 2mm will provide enough clearance to get around the supports.

This is a workaround I've used.  I've brought in a Support Blocker cube and set it to print as Normal Model.  It's 0.4mm thick and strategically placed so the tree support grows up from it.  The Per Model "Support Bottom Distance" is 0 so the bottom support interface glues to the blocker.
[image]

mikeywaldron, 2023-10-31:
I use klipper and z hops are still a bit dodgy with it, the avoid supports was on 0.6mm so I've increased this, also my initial layer height I've put up to counter act the loss of adhesion i get bed slingers etc... But they are part of the market so this should be taken into consideration more.

GregValiant, 2023-10-31:
_"bed slingers etc... But they are part of the market..."_
Remember though that UltiMaker does a lot of the heavy lifting on Cura and they don't make any bed slingers.  I wrote that post processor after my Ender threw a tall print on the floor.
I agree that tree supports in particular could use better adhesion.  Effective workarounds are needed because even if this request is acted on, my guess is that it will be a while.

GreetingsGQ, 2023-11-17:
There should be a setting for Support Brim Distance (default 0) separate from Brim Distance. Brim Distance did not affect the distance between the Brim and Supports prior to 5.3.