Ultimaker/Cura #17153: bug when opening cura

sd95100, 2023-10-30:
### Cura Version

4.13.0

### Operating System

rasperry os

### Printer

crealty ender 5

### Reproduction steps

i head successfully installed cura butt when i wanteed to install it on another system, it display "the plugin TrimesReader could not be loaded. Re-instaling the plugi might solve the issue" and "the plugin AMRReader could not be loaded. Re-instaling the plugi might solve the issue" I clicked on it but it did nothing, i restarted and i reintalled but that didn't do anything. on the old system it worked corerectly. and since then i can't import the file. 

### Actual results

it display "the plugin TrimesReader could not be loaded. Re-instaling the plugi might solve the issue" and "the plugin AMRReader could not be loaded. Re-instaling the plugi might solve the issue"

### Expected results

i hope it works

### Add your .zip and screenshots here ⬇️

[image]

[image]

sd95100, 2023-10-30:
I'm using a raspberry pi 400 and had to replace the sd card because it crashed. cura worked correctly but when i wanted to reinstall cura it caused this bug

fieldOfView, 2023-10-30:
You are likely missing a dependency: trimesh.

Ultimaker does not make a distribution of Cura that is compatible with Raspberry Pi OS. You are going to have to ask whoever packaged Cura for Raspberry Pi OS to fix the dependencies in their package.

closed 2023-10-30 by fieldOfView