Ultimaker/Cura #17155: Bed pre-heating doesn't match the initial bed temperature setting

ilko-k, 2023-10-30:
### Is your feature request related to a problem?

The build plate preheat temperature in the monitor tab uses the default bed temp. It's annoying to have to set it each time a print is restarted after a cancel. It should obviously use the "build plate temperature initial layer" instead because it comes before. Simple as that.

### Describe the solution you'd like

Prioritize the "build plate temperature initial layer" parameter when it is set manually.
[image]

### Describe alternatives you've considered

None

### Affected users and/or printers

All

ilko-k, 2023-11-03:
I see. So, which file do I have to edit to make this change in my local installation?

GregValiant, 2023-11-03:
Check the pull request and you will see what I did.

ilko-k, 2023-11-04:
I must miss something because I don't know how to do this without recompiling. There is no .qml file and I can't find such a string like "bed_temperature" in any file.

GregValiant, 2023-11-04:
I'm on Windows.  I made the change locally so I could debug it.  Both the affected files are text files and will open in Notepad or another text editor.  The file for the bed temperature is "PrintMonitor.qml" and is in:
"C:\Program Files\UltiMaker Cura 5.5.0\share\cura\resources\qml" 
Line 132 needs to be altered to have "material_bed_temperature_layer_0"

For the print temperature it's "ExtruderBox.qml" in:
"C:\Program Files\UltiMaker Cura 5.5.0\share\cura\resources\qml\PrinterOutput"
Line 25 needs to be altered to have "material_print_temperature_layer_0"

ilko-k, 2023-11-04:
Alright, thank you very much. I'm on Windows too but for whatever reason, my searches don't return expected results. When I type "monitor" for example, only files with the word "monitor" as prefix are showed...