Ultimaker/Cura #17156: multiply not separating builds on plate

dsc52359, 2023-10-30:
### Cura Version

5.5.0

### Operating System

windows 11

### Printer

ender 5 pro

### Reproduction steps

Place item to build on plate then attempt to multiply it.

### Actual results

Doesn't spread the builds apart correctly.

### Expected results

separation 

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2023-10-30:
In the Cura Preferences do you have "Ensure models are kept apart" checked?

dsc52359, 2023-10-31:
It is now thank you. I never had to do that before but thanks once again.

GregValiant, 2023-10-31:
I'll go ahead and close this then as it appears it was a setting issue.

closed 2023-10-31 by GregValiant

dsc52359, 2023-10-31:
Yes, please!